UtilitechAS/amsreader-firmware #15: Setup instructions

1v4r, 2020-02-01:
Hi. I tried to find some instructions on how to set this up. After flashing the image to an ESP8266 and rebooting it I detect the ESP wifi for a short period but I'm unable to connect to it. I tried to look for more instructions here and on @roarfred's repo but I was not able to find any more info.

gskjold, 2020-02-01:
Then there may be a problem in the code, causing it to reboot. What hardware are you using? And is it connected to the meters HAN port when this happens?

gskjold, 2020-02-02:
Tested with a clean board and got an error. Attached is a new firmware file.
[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/4143650/firmware.zip)

gskjold, 2020-02-07:
Assuming this was resolved

closed 2020-02-07 by gskjold