UtilitechAS/amsreader-firmware #19: Parity error on serial output that shows GUI IP adress

ArnieO, 2020-02-05:
The serial output that shows IP address for GUI is sent with NO parity, while (in my setup) debug data on serial output is sent with EVEN parity. 
Should be same parity.

gskjold, 2020-02-07:
In recently commited code, the parity is switched before debugging starts if meter type is set to Kamstrup.

gskjold, 2020-02-07:
This needs some further investigation. Possible to run different parity for RX and TX?

ArnieO, 2020-02-08:
I believe the original @roarfred code allows for different parity:
(This is from a modified *.ino file, but the marked lines have not been changed)
[image]

gskjold, 2020-02-14:
I've looked into this, and technically, you cannot have different parity for RX and TX on the same HardwareSerial.

I've come up with a hack though.. If DEBUG_MODE=1 it now sets up a SoftwareSerial with RX pin 3 with 2400 8E1/8N1 (depending on meter) and another SoftwareSerial with TX pin 1 with 115200 8N1 for debugging :)

closed 2020-02-14 by gskjold

ArnieO, 2020-02-14:
When you say "pin 1" and "pin 3" here, do you mean GPIO1 and GPIO3, which are the normal Rx/Tx pins on the ESP module?

gskjold, 2020-02-14:
Correct. Only activated with combination HW_ROARFRED=1 and DEBUG_MODE=1 in branch dev-v1.0.1.