UtilitechAS/amsreader-firmware #29: Not reading Kamstrup List 2

ArnieO, 2020-02-13:
According to NVE documentation (attached), the Kamstrup sends two lists: List 1 with 10 sec interval, List 2 with 1 hour interval. List 2 contains cumulative meter data. 
The current code (downloaded 12feb20) does not read List 2.
The kamstrup.h file indicates List1 and List3 (not List2), and both seems to collect cumulative data? 
This kamstrup.h file has previously worked correctly on my setup:
```
// Kamstrup.h

#ifndef _KAMSTRUP_h
#define _KAMSTRUP_h

enum class Kamstrup
{
	List1 = 0x19,
	List2 = 0x23
};

enum class Kamstrup_List1
{
	ListSize,
	ListVersionIdentifier,
	MeterID_OBIS,
	MeterID,
	MeterType_OBIS,
	MeterType,
	ActiveImportPower_OBIS,
	ActiveImportPower,
	ActiveExportPower_OBIS,
	ActiveExportPower,
	ReactiveImportPower_OBIS,
	ReactiveImportPower,
	ReactiveExportPower_OBIS,
	ReactiveExportPower,
	CurrentL1_OBIS,
	CurrentL1,
	CurrentL2_OBIS,
	CurrentL2,
	CurrentL3_OBIS,
	CurrentL3,
	VoltageL1_OBIS,
	VoltageL1,
	VoltageL2_OBIS,
	VoltageL2,
	VoltageL3_OBIS,
	VoltageL3
};

enum class Kamstrup_List2
{
	ListSize,
	ListVersionIdentifier,
	MeterID_OBIS,
	MeterID,
	MeterType_OBIS,
	MeterType,
	ActiveImportPower_OBIS,
	ActiveImportPower,
	ActiveExportPower_OBIS,
	ActiveExportPower,
	ReactiveImportPower_OBIS,
	ReactiveImportPower,
	ReactiveExportPower_OBIS,
	ReactiveExportPower,
	CurrentL1_OBIS,
	CurrentL1,
	CurrentL2_OBIS,
	CurrentL2,
	CurrentL3_OBIS,
	CurrentL3,
	VoltageL1_OBIS,
	VoltageL1,
	VoltageL2_OBIS,
	VoltageL2,
	VoltageL3_OBIS,
	VoltageL3,
	MeterClock_OBIS,
	MeterClock,
	CumulativeActiveImportEnergy_OBIS,
	CumulativeActiveImportEnergy,
	CumulativeActiveExportEnergy_OBIS,
	CumulativeActiveExportEnergy,
	CumulativeReactiveImportEnergy_OBIS,
	CumulativeReactiveImportEnergy,
	CumulativeReactiveExportEnergy_OBIS,
	CumulativeReactiveExportEnergy
};

#endif
```

gskjold, 2020-02-13:
The lists in the current definitions in Kamstrup.h are for 1phase and 3phase respectively, and both cover the whole range of the largest available list. Which part of the list is used, is determined by the list size and implemented in HanToJson.cpp. I will have a closer look at this in the near future.

gskjold, 2020-02-13:
Compared the attached documentation to the code, and it looks correct.
I have done two changes to pursue this:
- Increased limits for JSON and MQTT
- Added list size in JSON as "ls". This should read 25 for short list and 35 for long

See branch dev-v1.0.1
Attached fw for hw_roarfred
[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/4200693/firmware.zip)

ArnieO, 2020-02-14:
I confirm: With branch dev-v1.0.1 I now receive Kamstrup List 2.

closed 2020-02-14 by ArnieO