UtilitechAS/amsreader-firmware #31: Local production (e.g. solar panels) in GUI

ArnieO, 2020-02-14:
Whenever Import Power is zero and Export Power positive (e.g. while solar panels produce more than the self-consumption), the GUI meter shows nothing:
[image]
Improvement options:

1. In the GUI Configuration, make an option to provide local production installation kWp (kW peak) power, and let the meter scale go from minus [Installed_kWp] to plus [main fuse power]. 
(And if possible green meter dot when delivering to grid, red dot when consuming from grid.)
2. Beside the "Consumption" meter, a separate meter "Production" that is non-zero when delivering to grid.

gskjold, 2020-02-14:
Funny, I already have a uncommitted local change with "productionCapacity" as a configuration option with this in mind :)

I like the idea of combining the two graphs, but not sure if the library is easy to configure for this, but we'll see what I can come up with.

ArnieO, 2020-02-14:
Great minds think alike...  ;-)

gskjold, 2020-02-15:
Ended up being a separate graph... Maybe combine them in the future

closed 2020-02-15 by gskjold