UtilitechAS/amsreader-firmware #32: ESP Vcc

ArnieO, 2020-02-14:
The unit for the figure returned by EDP.getVcc() is 1/1024 V
So to get the 'correct' voltage in V you need to divide by 1024. 
Like:
`float vccVolt = ((float)ESP.getVcc())/1024;`

closed 2020-02-15 by gskjold

ArnieO, 2020-02-22:
In the MQTT message Vcc has six decimals while the measurement resolution is 1/1024 V.
So three decimals is sufficient / more correct - and will reduce the length of the MQTT message.

ArnieO, 2020-02-23:
I suspect Vcc in the GUI is rounded to one decimal but presented with two decimals. I followed the GUI closely while MQTT value changed by 0,02 V but last decimal in GUI was always zero.

gskjold, 2020-02-23:
Does not happen on any of my devices. This could be browser specific. Will change MQTT message :)

ArnieO, 2020-02-23:
I have not yet seen the second Vcc decimal in GUI different from zero, on Safari (iPad) and on Firefox (PC).