UtilitechAS/amsreader-firmware #34: Only send to MQTT if data changed

gskjold, 2020-02-17:
Add some code to only send if the values have changed. 
Motivation: Preserve power. Not everyone have a low traffic, low DTIM network.
Maybe it doesn't matter much? 
Maybe optional thresholds makes this a valid enhancement?

Needs testing

ArnieO, 2020-02-17:
With full accuracy of P, Q, Po, Qo, 3 * voltage and 3 * current, I think the probability is very low that all values in two sequential MQTT-messages would be identical.

gskjold, 2020-02-17:
I guess that will never happen for the 10s list, but for the 2.5s maybe? Using power every 10s vs 2.5s does some impact. For my Aidon meter providing 30mA(?) it isn't too much of a problem, but I am currently planning for testing this on Kaifa with 20mA(?) which may need som tweaks.

Also, if there is multicast traffic on the network, this seriously impacts the power consumption.

Maybe have a option to skip the 2.5s update if the power is too low.

gskjold, 2020-03-10:
Only applied when using raw payload with MQTT

closed 2020-03-10 by gskjold