UtilitechAS/amsreader-firmware #35: Uptime rollover after 50 days

ArnieO, 2020-02-17:
ESP uptime is direct readout of millis(), which has data type **unsigned long**, ranging from 0 to 4,294,967,295.
So it will roll over to zero after approximately 50 days.

gskjold, 2020-03-08:
Fixed for UI.
JSON does not like uint64, so adding this as millis uptime in json is not working. Best I can come up with is using uptime in seconds. Or just accept that "up" in JSON to MQTT is rolling over.

ArnieO, 2020-03-09:
I think the expanded number could be sent as string in the JSON. But as this timer is not synchronized with anything I do not really see the relevance of millisecond resolution.
So I agree with you: Uptime in seconds should be fine.

gskjold, 2020-03-09:
Agreed, milliseconds resolution is excessive in this case. It would also require the receiving part to support uint64.

Reducing resolution on "up" field to seconds and announcing this as an API change in next release.

closed 2020-03-10 by gskjold