UtilitechAS/amsreader-firmware #37: Web GUI: Change "Production" to "Exported power" (or a better term!)

ArnieO, 2020-02-18:
The name of the second meter ("Production") is misleading. 
This is not the production but the exported power, which is (Production - SelfConsumption).

I am not sure which is the best / most intuitive term to use here.
- Exported power ?
- Delivery to grid (and change the first meter to "Consumption from grid") ?
- Delivery ?
- (Any other ideas ?)

gskjold, 2020-02-24:
Leaning towards Import/Export when you have production, as "Consumption" doesn't really feel right either. Sticking with Consumption when no production is configured.

Commited change

ArnieO, 2020-02-24:
OK, this makes sense. 
- Import/Export is a commonly used term, so should not be confusing for those having configured production.
- Sticking with Consumption when no production is configured is also fine.

closed 2020-02-24 by gskjold