UtilitechAS/amsreader-firmware #40: Add temperature to web interface

Pecillo, 2020-02-23:
Feature request

Add another gaugemeter with temperature display.

gskjold, 2020-02-24:
In the upcoming v1.1.0, we have added temperature to the dashboard, but only as text. I cannot think of a reasonable scale for a gaugemeter that would apply well to temperature in a fuse box, but I am open for suggestions.

[image]

Pecillo, 2020-02-24:
That's great!

I was accessing the webpage from my computer and I had tons of free space. But from a mobile... Well I agree that it might be  bit tight to squeeze in another graph.

closed 2020-02-24 by Pecillo

gskjold, 2020-02-24:
That is also true, there is a bit of extra space on a desktop :) In upcoming release there will also be a extra gauge if you have local production (ex. solar power) which takes up some extra space as well

I have launched a pre-release of v1.1 today:
https://github.com/gskjold/AmsToMqttBridge/releases/tag/v1.1-rc1

This is possibly a unstable build, only use if you want to test new features.

Pecillo, 2020-02-24:
I'm a bit ahead of schedule... I'm still waiting for the power company to replace my meter. But I had the electronics finished this weekend.
But I'm happy to see that at least my local provider here in Sweden followed the standard that is used in Norway.

gskjold, 2020-02-24:
Neat! That's good to know. Is the data format all the same?

Pecillo, 2020-02-24:
I'll get a Aidon meter, lots of the documentation I got from Härryda Energi was Norwegian.

Pecillo, 2020-02-24:
I posted this in "Svenska hemautomatiseringsgruppen" yesterday evening:

> Nu är helgens lödövning klar, en mqtt brygga till elmätaren.
> Då saknas bara att Härryda energi kommer och byter vår mätare.
>
> Hårdvaran är roarfreds:
>
> https://github.com/roarfred/AmsToMqttBridge
>
> Och en fork av koden med stöd för Aidon mätaren vi kommer få i Härryda:
>
> https://github.com/gskjold/AmsToMqttBridge
>

A group with close to 37k members...

gskjold, 2020-02-24:
> I'll get a Aidon meter, lots of the documentation I got from Härryda Energi was Norwegian.
That is a good indicator it will be the same. Regardless, I'm sure we can make adjustments if necessary.

> I posted this in "Svenska hemautomatiseringsgruppen" yesterday evening:
> A Facebook group with close to 37k members...
I guess  there is a chance that interest in this code will pick up then  :smile:

Pecillo, 2020-04-28:
Quick update now that I have got my meter replaced... Works perfectly with the Aidon meters installed by Härryda kommun. I know that Ale kommun is installing the same model as Härryda.

gskjold, 2020-04-29:
Thanks for the update, good to know that it actually is all the same in Sweden.