UtilitechAS/amsreader-firmware #41: WEB GUI responds incorrectly to net power export.

ArnieO, 2020-02-24:
When my solar panel kicks in and I have net power export the GUI signals HAN error (red indicator), and no export data is indicated:
[image]
MQTT messages are still flowing, and my Node-Red implementation shows export power correctly.
As soon as net import is back, the red HAN light goes green, and the GUI behaves correctly, showing imported power.

gskjold, 2020-02-24:
https://github.com/gskjold/AmsToMqttBridge/commit/9671e1eba334a1d7b052175086f43c2f602508a7

Fixed in this commit. Maybe I also fixed the Vcc precision?

ArnieO, 2020-02-24:
Great - it works correctly now! For the Vcc second decimal we will have to wait and see, the small voltage spikes occur very randomly. 
[image]

closed 2020-02-24 by ArnieO