UtilitechAS/amsreader-firmware #49: Connected to MQTT! loop

vsurlien, 2020-03-16:
I have tried your code, and somtimes it workes fine for a few hours and then it stops and goes into a loop, see picture. Sometimes it goes directly into this loop.

[image]

[image]

Do you have any idea of what is wrong, or what to check?

I have other traffic going to the same broker, but it is stable.

I also have been using the orginal roarfred code with the PubSubClient.h and it worked fine, just that i was not able to get kwh readings out.

gskjold, 2020-03-16:
Hard to tell what this could be. A few questions to make my search easier:
Is the loop constant? Or is it reconnecting every 5 seconds like it is supposed to?
Have you checked the logs on the MQTT server for clues?
Does the web interface show an error message for MQTT?

vsurlien, 2020-03-17:
I was getting a lott of these messages:

> 1584400347: New connection from AMS2MQTT_D1 on port 1883.
1584400347: Client AMS2MQTT_D1 already connected, closing old connection.
1584400347: New client connected from AMS2MQTT_D1 as AMS2MQTT_D1 (p2, c1, k10).

So I tried to setup a new MQTT broker on a different computer. It has been working all night now without a without a glitch. It seam like it is a problem with the "orginal" broker. This is strange, since i have other traffic going there without any problem.

Anyway, it seams like it is a broker issue and not a issue with your/roarfred code.

closed 2020-03-17 by vsurlien

gskjold, 2020-03-17:
OK, normally you get this message if another client connected to the broker has the same client id

vsurlien, 2020-03-23:
The D1 mini unit died just a few days after. So HW failure was the real problem. Thanks for your support.