UtilitechAS/amsreader-firmware #51: HAN port data format changed

kagjerde, 2020-03-22:
Hi, thanks for keeping this project alive, and I really like your new WEB GUI.

I had version 0.2.0 working in January and February, using Kaifa meter. 
Utility company is BKK in Bergen (where I live).

Everythink worked until resently, when I noted at the data as corrupt.
I used the AMS bridge to MQTT and use a NodeRed flow to extract the data into Home Assistant.

Now I do not have any valid data, I have checked Mosquitto and it seems the data from the meter have changed. 

Tried to upgrade to your 1.1.0 version, but here I have no data at all. All "buttons" are green except HAN bus (red), but I have access to the WEB GUI. 
There are no data sent to MQTT either.

I have two questions:
- Can you include a "dubug" enable in the WEB GUI ? it would make it easier to enable?
- Or possibly a "transparent" mode where the raw HAN data is simply passed to MQTT? 
  In version 0.2.0 I have MQTT data, but the format looks different.
- Are you aware of any protocol changes executed by any of the power companies lately?

Thanks in advance.
Kjell

gskjold, 2020-03-23:
Good to hear you like the continued work on this project! The new code have been properly tested on both Aidon and Kamstrup, but I am not aware of any extensive testers that have a Kaifa meter. Since HAN status is red in the UI, it suggest that the code is either not receiving or is unable to interpret the received data. I will compare the code between 0.2.0 and 1.1.0 when I have some time, just to be sure nothing changed for Kaifa. What hardware are you using? One or three phase distribution?

1. This will be added in the upcoming version: https://github.com/gskjold/AmsToMqttBridge/issues/7
2. This is a great idea! Will look into this once I am finished with my current task.
3. I have not been informed of any change from anyone, no.

gskjold, 2020-03-24:
Remote debugging resolved. Unable to send raw data to MQTT due to the length of the string. Not too keen on extending the MQTT buffer size just because of that. But the data will be printed in debugger, which will now be available via telnet if enabled in UI.

closed 2020-03-24 by gskjold

kagjerde, 2020-03-26:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi again,

For your information, the release 1.1.1 fixed the problem, I have not tested 1.1.0 but it works now using Kaifa meter.

The problem may have been on my side, so please disregard the failure report for now.

Best Regards
Kjell

gskjold, 2020-03-26:
Thanks for the update, that is great to hear! v1.2.0 includes remote debugging and will be released in a couple of weeks probably.