UtilitechAS/amsreader-firmware #52: MQTT topic from "case 3" is not transfered

Allram, 2020-03-23:
Hello,

Thanks for keeping this project alive!
I have Aidon 3Phase IT meter (6525) and i have this running on a Wemos D1 mini with MQTT to my main server.

I get all of the values defined in case 1 and 2 in the HanToJson.cpp but the values from case 3 is not transfered. Have you encountered this?

This is the values that are not beeing sent on the MQTT:
      ```
  case 3:
            jd["rtc"]  = data.getMeterTimestamp();
            jd["tPI"]  = data.getActiveImportCounter();
            jd["tPO"]  = data.getActiveExportCounter();
            jd["tQI"]  = data.getReactiveImportCounter();
            jd["tQO"]  = data.getReactiveExportCounter();
```

gskjold, 2020-03-24:
Glad to hear you like it! You should receive list3 at 10s past the hour. I just checked now and mine sent the values at that time (Aidon 6534). I looked through the code for IT meter, but cannot see a reason it shouldn't work. The only reason I could think of is that either the MQTT buffer size is too small for your list3 JSON length. It is set to 384, and my list3 is 305. Seems weird that yours would be 80 chars longer than mine though... If you are compiling the code yourself, try setting the MQTT buffer to 512 here:
https://github.com/gskjold/AmsToMqttBridge/blob/30d73c3a6e32207475ad3855d06b0691be6772b2/src/AmsToMqttBridge.ino#L48

Allram, 2020-03-24:
Did a recompile with MQTT buffer to 512, but no change in the received data.

here is the data i receive every 10s
{"id":"98:F4:AB:DA:66:E8","up":59438,"t":0,"vcc":3.327,"rssi":-74,"data":{"lv":"AIDON_V0001","id":"7359992902654080","type":"6525","Q":0,"PO":0,"QO":781,"I1":6.3,"I2":0,"I3":5.7,"U1":235.7,"U2":234.2,"U3":233.5,"P":1995}}

Allram, 2020-03-24:
Problem solved with increasing the MQTT buffer to 512.
Do you want me to do a PR, or do you fix it yourself? :)

gskjold, 2020-03-24:
Cool! Feel free to PR :) I will have it tested with some other setups

Allram, 2020-03-24:
Closed with https://github.com/gskjold/AmsToMqttBridge/pull/54

closed 2020-03-24 by Allram