UtilitechAS/amsreader-firmware #55: Changed string format for tPI

kagjerde, 2020-03-30:
I am using NodeRed to link the data into Home Assistant, and noted that you changed the string format recently, but do not know exactly when:

String provided in February 2020:
""temp":18.5,"lv":"KFM_001","id":"6970631402281XXX","type":"MA304H3E","P":6947,"Q":81,"I1":20.558,"I2":13.881,"I3":25.14,"U1":210.6,"U2":0,"U3":202.6,"tPI":103184047,"tPO":0,"tQI":4831895,"tQO":255935}}"
Please note that the beginning of the string is truncated.

String Provided in version1.1.1:
"{"id":"EC:FA:BC:4C:XX:XX","up":274830607,"t":1585533610,"vcc":3.437,"rssi":-47,"temp":22.87,"data":{"rtc":1585533610,"tPI":1.134553e8,"tPO":0,"tQI":5017038,"tQO":303751,"lv":"KFM_001","id":"6970631402281XXX","type":"MA304H3E","Q":0,"PO":0,"QO":327,"I1":20.93,"I2":21.829,"I3":29.34,"U1":215.5,"U2":0,"U3":213.3,"P":8834}}"

I have re-structured most of the data, as the "temp" reading is now handled differently. But this is not a problem.

However the data for "tPI" is a problem, I use this data to calculate aggregated power consumption and link this to price from nor-pool to calculate price per hour.

In the earlier versions the full string was given (103184047), and now only (1.134553e8).

Can you please consider giving the full figure with all digits ? 
Please note that capturing this figure is a bit of a hassle, since the reader only prints it once every hour.

I am using the original Roar Fred HW. So I have used the HW-01 version of the image.
Meter is Kaifa.

gskjold, 2020-03-30:
Turns out that this is a default behaviour from ArduinoJson. It is possible to change, so I will do that.

Curious, have you actually used 10M kwh? Or is the value from Kaifa i w? On the two other meters I have standardized the output to kwh, and I aim to do the same for Kaifa as well

gskjold, 2020-03-30:
Changed limit for scientific annotation in v1.1.2.
Judging from the document on Kaifa, the accumulated value is given in watts, this will be converted to kwh in JSON from v1.2.

closed 2020-03-30 by gskjold

kagjerde, 2020-03-30:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi again,

The figures are the ones from the meter, I think its in W but KWh is ok.

Kjell