UtilitechAS/amsreader-firmware #64: Negative current

stenjo, 2020-04-16:
When production and exporting energy the current value is displayed incorrectly. 
[image]

stenjo, 2020-04-17:
Trying to compile a local version to test this. Will soon be back with test results.
[image]

gskjold, 2020-04-17:
Thanks, while your PR probably fix it, I can see a number of other things wrong with the surrounding code that will need to be addressed as well to avoid future problems. I think the best way will be to take the code apart and do type casting for each of the types. I have some ideas on best approach, I'll see if I can whip up a quick solution later.

stenjo, 2020-04-17:
Seems like the fix removed the bug.
[image]

stenjo, 2020-04-18:
[image]
Still looks good.

gskjold, 2020-04-18:
I did some work on it this morning where I corrected the conversion for int8, int16 and uint32. Please see this branch: https://github.com/gskjold/AmsToMqttBridge/tree/dev-v1.2.3
I've run a number of tests on the code and it gives correct results.

stenjo, 2020-04-19:
Initial test looks good with your revised dev-v1.2.3
[image]

gskjold, 2020-04-19:
Great! I also had it running in parallel with 1.2.2, can't see it changed anything on the other values either.

closed 2020-04-19 by gskjold