UtilitechAS/amsreader-firmware #67: Provide export data when not changed

stenjo, 2020-04-20:
Seems like export data (part of the list sent every 10 seconds) is only published if it differs from the previous value. This causes data to be displayed incorrectly when smoothing the curves.
In the image below the value is actually zero up and until 10 seconds before the next non-zero value. As these datapoints are not transmitted the average value is not calculated right.
As the list is sent every 10 seconds the zero values should be sendt even if they are the same as the previous value
[image]

Suggest a feature that enables adding this as an option somewhere - in case this causes too much traffic for some...

gskjold, 2020-04-21:
Good idea! We'll make this an optional feature.

stenjo, 2020-04-21:
I could make an PR on this if we agree on how this should look. For now I have modified my own code to get all values I need

stenjo, 2020-04-21:
I could make an PR on this if we agree on how this should look, where the settings should be and what range of options we should go for. For now I have modified my own code to get all values I need.

gskjold, 2020-04-22:
I can think of two possibilities here.

1. Duplicate the RAW option in the dropdown:
1=RAW (minimal)
2=RAW (full)
Or similar naming. Then change:
```
if(lastMqttData.getMeterId() != data.getMeterId()) {
```
to
```
if(lastMqttData.getMeterId() != data.getMeterId() || config.getMqttPayloadFormat() == 2) {
```
etc..

2. Tick box under MQTT config page with corresponding new variable (ex. bool mqttOnlySendChanged) in config object. And change to:
```
if(lastMqttData.getMeterId() != data.getMeterId() || !config.isMqttOnlySendChanged()) {
```
etc..

Bump EEPROM_CHECK_SUM in AmsConfiguration.h when adding config parameter, add a loading function for the new version, add in load() and hasConfig(). Also modify save() with the new parameter.

I'm sort of thinking the first option is as good as any in this case, less coding, reusing existing config parameter and less cluttering av user interface, whats your take?

Names of payload format and variable naming only for example, final naming up for discussion.

stenjo, 2020-04-22:
I'd go for option 1. Mostly because of simplicity. Downside is that full reporting only would be available on RAW (unless one also added JSON (full) and JSON (minimal)) but that might be just sufficient.

gskjold, 2020-04-22:
True, but right now all values are always included in JSON, so I think we'll just add a minimal version of that in the future, if requested.

PR #68 (2020-05-01): Feature/fullreport
This change imlements a RAW (full) reporting option that will send all data to MQTT even if they have not been changed since the last report was sent.
Fixes #67
closed 2020-05-01 by gskjold