UtilitechAS/amsreader-firmware #74: dev-v1.3.0: Wifi configuration on esp32 fails.

atlej68, 2020-05-07:
made a test build, but are not able to pass WiFi_connect() after configuration (dhcp). 
esp32 stays in AP-mode. 

SSID and key printed on debug console looks fine.

debug messages: 
␛[0m(I) (WiFi_connect)(C1) Connecting to WiFi network: <MYSSID>
␛[0m[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 9
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 118
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 9

gskjold, 2020-05-07:
This is a development branch and not ready. These issues will be sorted during first stage of testing

gskjold, 2020-05-08:
Dang the close and comment button! Anyways, commited some changes that works for both my d1 mini and my d32, so how does this work for you? Judging from the error messages I sort of doubt it changes anything, but who knows. 

Can you tell me anything about the wifi configuration, are you using static IP? If so, is DNS added?

atlej68, 2020-05-08:
No worries, not in a hurry - just curious, have a stable version based on 1.2.3 I am running now. 

Wifi connection setup seems to work now!   The wifi config I tested yesterday is the simplest possible: only  ssid and psk(wpa2-personal) input. Running towards an ASUS Router. 

Never tested static ip yesterday ( asus/merlin lets me assign ip to the devices even if they connect via DHPC.)   - did test now and it works as well. 

The error message above is only there in AP mode, seems to be related to dns?   Stops when it connects to my network now.

gskjold, 2020-05-09:
Aha, that makes sense, this is the NTP client I had to add to make SSL work. I should disable that when in AP mode.

gskjold, 2020-05-10:
Should have eliminated the error messages in AP mode as well now. They might show up in station mode, but nothing to worry about.

closed 2020-05-10 by gskjold