UtilitechAS/amsreader-firmware #80: Fix building conflict with bas64

gskjold, 2020-06-06:
The base64 lib currently use causes building on Windows to crash. Look into using already available code from base espressif

ArnieO, 2020-07-24:
The cause for this seems to be that the Windows platform cannot distinguish between the libraries **Base64** (capital "b") and **base64**. 
HTTPClient.cpp use `#include "HTTPClient.h"`
Your code in AmsWebServer.cpp use `#include "Base64.h"`

We need a workaround to avoid this confusion.

ArnieO, 2020-07-24:
See also this comment in a closed issue in the **Base64** library:
https://github.com/agdl/Base64/issues/1#issuecomment-309395846

It seems like this library should not be used for ESPs.

gskjold, 2020-07-25:
Thanks for the info! I have now switched to the internal espressif base64 library, so that we will not have any build conflicts.

closed 2020-07-25 by gskjold