UtilitechAS/amsreader-firmware #81: Failing when building on newest release using VScode and PlatformIO

Allram, 2020-06-13:
Hi,

I get this error message when trying to build the latest release, any tips? 

`ModuleNotFoundError: No module named 'css_html_js_minify':
  File "C:\Users\Vegard\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 176:
    env.SConscript(item, exports="env")
  File "C:\Users\Vegard\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Vegard\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Vegard\Desktop\AmsToMqttBridge-master\scripts\makeweb.py", line 4:
    from css_html_js_minify import html_minify, js_minify, css_minify`

gskjold, 2020-06-21:
I totally forgot about this during last dev cycle. You have to do a `pip install css_html_js_minify` to have the necessary python modules to minify web content. I will make this optional feature as soon as I have time to sit down. Just home from holiday, so I have a few things in the pipeline right now.

gskjold, 2020-07-06:
Now the script only uses minifier if available.

closed 2020-07-06 by gskjold