UtilitechAS/amsreader-firmware #86: Support for P1-wifi gateway

fn87, 2020-08-27:
I have an: http://www.esp8266thingies.nl/wp/ that I would like to use with this project.

I can flash it but I do not get further than that. I cannot get it to boot in to AP and serial debug just gives me strange data.

I would appreciate any pointers to get it up and running.

gskjold, 2020-08-28:
If I understand the link correctly, this is a shield for Wemos D1 mini. I would disconnect the shield before flashing and setting up the D1.

As for the debugger, because the default is to use UART0 for MBUS, the default port settings is 2400 8E1, this should give you some output at least.

gskjold, 2020-09-05:
Closed due to inactivity

closed 2020-09-05 by gskjold

fn87, 2020-09-05:
Thanks for your response. I decided on coding something myself based on ESP Mega as it boots with the shield without issues. Code is something I know, hardware...not so much.