UtilitechAS/amsreader-firmware #89: Add ADC configuration for vcc and analog temperature

gskjold, 2020-09-05:
Add possibility to configure resistor values for the voltage divider for vcc.
Analog temperature needs configuration for voltage at zero as well as increase rate (mV/deg)

mikkel75, 2021-09-27:
So much looking forward to this feature...
BTW, I have created a ESP8266 HAN module for the Kamstrup Omnipower. Is it possible to get it included under Hardware? :)

gskjold, 2021-09-30:
I only saw the first line in the notification, sorry about that. Please describe your hardware in a new issue and I will have a look at it.

mikkel75, 2021-10-01:
Hi,

I have made a module that plugs directly in to the Kamstrup Omnipower as a HAN module.
It is based on the ESP8266.

It looks like this:

***@***.***

It has UART RX on GPIO13 (TX on 15).
The ADC is measuring the battery voltage (before the DC/DC) with a resistor divider ADCin = 33/133 *Vbat. Vbat can be up to 4.4V

BR,
Mikkel

mikkel75, 2021-10-01:
Oh and one more thing – could you please add support for UART2 on the ESP8266 with the swap command?

Something like this…

void setupHanPort(int pin, int newMeterType) {
          debugI("Setting up HAN on pin %d for meter type %d", pin, newMeterType);

          HardwareSerial *hwSerial = NULL;
          if(pin == 3) {
                    hwSerial = &Serial;
          }

          if(pin == 13) { // GPIO13
                    hwSerial = &Serial;
                    Serial.swap(); // change UART0 pins to GPIO13 and GPIO15
          }

BR,
Mikkel

gskjold, 2021-10-03:
This module you have made, is that commonly available to others?

Regarding UART2, I will have a look at it for next release.

mikkel75, 2021-10-03:
It will be released for free on github shortly...just have to get the last debug done. I have proto working.

-------- Oprindelig besked --------

Dato: 03.10.2021 09.43 (GMT+01:00)
Til: gskjold/AmsToMqttBridge ***@***.***>
Cc: mikkel75 ***@***.***>, Comment ***@***.***>

mikkel75, 2021-10-05:
The board will be available here:

https://github.com/mikkel75/KamstrupHAN

gskjold, 2021-11-23:
Note to self, do not reset vcc config on factory reset

ArnieO, 2021-11-23:
> Note to self, do not reset vcc config on factory reset 

Maybe included in your comment, but should also not be reset when reconfiguring wifi via Access point.

closed 2021-11-26 by gskjold