UtilitechAS/amsreader-firmware #94: Slow down the amount of updates sent via MQTT?

Allram, 2020-10-26:
Hello,

Is there a way to slow down the amount of updates that are being sent to the MQTT server? Like every 10 or 30seconds instead of every 2.5seconds?

My Athom Homey is struggling with MQTT updates every 2.5seconds (high cpu load).

gskjold, 2020-10-27:
Hi,
Sorry, but there is currently no such feature. I am curious though, do you have a heavy set of rules that runs every time you receive data? I find it strange that anything would struggle with 2.5s update rate.

Allram, 2020-10-27:
I have alot of rules 😂
1 for each phase with Amps
1 for each phase with voltage
1 for Import power
1 for export power
1 for kWh
And they each trigger additional virtual devices and real devices depending on value.

Solved it mostly by deactivating the rules and activate them again every 30sec, at least that made the device usable again. 👌

closed 2020-10-27 by Allram

ArnieO, 2020-10-28:
@Allram One option could be to use Node-RED, and let it average a number of incoming values before passing them on to your system.

Allram, 2020-10-28:
> @Allram One option could be to use Node-RED, and let it average a number of incoming values before passing them on to your system.

Nice! Will give that a try. Thanks.

ArnieO, 2020-10-30:
I am not a Node-Red expert, but have found the Aggregator node to be quite interesting/useful: 
[image]
[image]