UtilitechAS/amsreader-firmware #97: Improve how we configure timezone

gskjold, 2021-01-10:
Currently it is saved in seconds offset for both standard time and summer time. Change this to either a choice of CET, CEST etc or Europe/Oslo etc. Problem with the latter is that if Norway decides to drop summer time, we will have to update the code.

Reason for wanting the change is to make it easier to take timezone into account when presenting data to end user.

gskjold, 2021-01-20:
configTzTime("CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00", ntpServer); 

https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html
https://remotemonitoringsystems.ca/time-zone-abbreviations.php

aadnehovda, 2022-09-03:
If the meter provides the current local time then is it necessary to even consider time zone or query NTP in the first place? Maybe kWh aggregations per day/hour will be more accurate too (match the power company's reports) if you always use the same clock as the meter?

Otherwise it should be possible to automatically pick up the local time zone and get the current UTC offset by querying a GeoIP service like https://ip-api.com e.g. once every hour.

gskjold, 2022-09-05:
First of all, aggregation per hour/day is based on the meters hourly report, so it would be pretty weird if it does not match with your grid operators report.

As for the timezone:
- Not all meters report date and time.
- The different meters does not report time and date in a consistent way, so using that would cause more problems than any other solution.
- SSL communication (ENTSO-E API) requires exact time to work, making NTP necessary.

bardahlm, 2022-09-06:
You can get timezone from the client, see https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Intl/DateTimeFormat/resolvedOptions#examples.

closed 2022-11-24 by gskjold