UtilitechAS/amsreader-firmware #98: Test v1.5-rc1

gskjold, 2021-01-21:
Changes:
- Major code rewrite to optimize memory usage
- Improved user interface
- Using built-in BearSSL for encrypted meters for ESP8266, removing mbedtls
- Removed support for loading config from v1.2.x series firmware
- Fixed security check for HTTP POST methods
- Fixed timestamp from DLMS with timezone and DST

Added:
- Support to retrieve energy price from ENTSO-E API (ESP32 only, API key required)
-----
Help wanted to test some of the following features (Do not except all to be tested)

Test on following hardware:
- [ ] roarfred hardware
- [ ] Generic esp8266
- [ ] Generic esp32

Test upgrade from:
- [ ] v1.3
- [ ] v1.4

Test on meters:
- [ ] Aidon
- [ ] Kaifa
- [ ] Kamstrup
- [ ] Omnipower

Test functionality of:
- [ ] Meter configuration with and without fuse size and production capacity
- [ ] WiFi with and without static IP
- [ ] MQTT with and without SSL
- [ ] Web security
- [ ] NTP configuration (indication if it is working in "Prices" section)
- [ ] ENTSO-E API (Only working for ESP32)
- [ ] Firmware upgrade
- [ ] Factory reset

Test MQTT payloads:
- [ ] Json
- [ ] Raw
- [ ] Domoticz

Allram, 2021-02-15:
> Test on following hardware:
> 
>     * [ ]  roarfred hardware
> 
>     * [x]  Generic esp8266 - Wemos D1 Mini
> 
>     * [ ]  Generic esp32
> 
> 
> Test upgrade from:
> 
>     * [x]  v1.3
> 
>     * [x]  v1.4
> 
> 
> Test on meters:
> 
>     * [x]  Aidon
> 
>     * [ ]  Kaifa
> 
>     * [ ]  Kamstrup
> 
>     * [ ]  Omnipower
> 
> 
> Test functionality of:
> 
>     * [x]  Meter configuration with and without fuse size and production capacity
> 
>     * [x]  WiFi with and without static IP
> 
>     * [x]  MQTT with and without SSL
> 
>     * [x]  Web security
> 
>     * [x]  NTP configuration (indication if it is working in "Prices" section)
> 
>     * [ ]  ENTSO-E API (Only working for ESP32)
> 
>     * [x]  Firmware upgrade
> 
>     * [x]  Factory reset
> 
> 
> Test MQTT payloads:
> 
>     * [x]  Json
> 
>     * [x]  Raw
> 
>     * [x]  Domoticz

Almost everything seems to work on Wemos D1 mini, except i can't get into Temperature settings in WebUI. It says 'Error loading data'.

But if i add temperaturesensor in v1.4 and then upgrade to 1.5-rc1 it can read the sensor and publish it by MQTT, but i still can't access the temperature settings in WebUI.

gskjold, 2021-02-18:
Alright, thanks! I will look into the temperature settings

gskjold, 2021-03-10:
I can't figure out why temperature settings is not working for you. Could you open http://<ams_ip_or_host>/temperature.json and copy/paste here? There must be something in the output that is not right.

gskjold, 2021-03-10:
I have tested on several devices now and can't find anything wrong. Releasing v1.5.0 as is and closing this issue. Please report a new issue with details from temperature.json if the problem still persists with the binaries from the latest release.

closed 2021-03-10 by gskjold

Allram, 2021-03-10:
Works with latest release. Seems like temperature.json wasn't included in my build..