UtilitechAS/amsreader-firmware #100: error saving configuration

gittela, 2021-02-21:
Er det noen begrensninger i hva slags generisk 8266 man bruker? Jeg forsøker å får dette opp på en esp-03 med 1 meg flash. Den kommer opp som aksesspunkt men vil ikke lagre config. Får kun feilmelding " error saving configuration". Jeg har brukt det ferdige imaget, ikke bygget nytt selv.

ArnieO, 2021-02-21:
Er du sikker på at din ESP-03 har 1MB flash? Lenge siden jeg har brukt ESP-03, men mener å huske 512k? 

Uansett tenker jeg du stanger i taket på flash. Denne er primært beregnet for ESP-12 (eventuelt moduler med den på) som har 4 MB.

gittela, 2021-02-21:
Fant ut av det. Den mangler noen i/o så den vil aldri funke til dette, men kan godt være at 1MB er i minsta laget ja (den har 1 MB). Nå har jeg litt mer issues med mbus-adapteret og forsøker å få telnet-debugging opp.

closed 2021-02-21 by gittela