UtilitechAS/amsreader-firmware #104: Device is bricked when attempting to connect to Azure IoT Hub as an IoT device

ofredstie, 2021-04-05:
When I configure MQTT on my [device](https://github.com/roarfred/AmsToMqttBridge/tree/master/Electrical/HAN_ESP_TSS721) to [connect to Azure IoT Hub](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly-as-a-device) it disappears from my local network. 

I noticed on one occasion that parts of the Password string appeared in the Username field before I lost connection with the device. 

Username: {iothubhostname}/{device_id}/?api-version=2018-06-30
Password: SharedAccessSignature sig={signature-string}&se={expiry}&sr={URL-encoded-resourceURI}

gskjold, 2021-04-06:
What version are you running? First thought was overflow of a field, but I kind of doubt it with what you are describing. It sounds more like a byte shift in the reading.

Current limitations:
host: 128
clientId: 32
topic: 64
username: 64
password: 64

Let me know if you overflow any of these fields

ofredstie, 2021-04-06:
I'm using the latest version, Release v1.5.3

gskjold, 2021-04-06:
Have you checked if any of your configuration input overflow the number as characters as specified in my last post? GUI should prevent overflow, but this could depend on browser.

ofredstie, 2021-04-06:
Host: 38
ClientId: 8
Topic: 8
Username: 64
Password: 144

ofredstie, 2021-04-06:
Will this help?

`struct MqttConfig {
	char host[128];
	uint16_t port;
	char clientId[32];
	char publishTopic[64];
	char subscribeTopic[64];
	char username[128];
	char password[256];
	uint8_t payloadFormat;
	bool ssl;
}; // 420`

gskjold, 2021-04-07:
You would also have to move the EEPROM address for the struct:
https://github.com/gskjold/AmsToMqttBridge/blob/606bac100a4c605f91224a02a01a00ce36b2af6b/src/AmsConfiguration.h#L13
F.ex. 1024. Erase EEPROM and reconfigure the device from scratch.

I have to ask though, why the ludicrous username and password length? Surely 64 should be more than enough to secure the connection.

I'm trying to reduce the memory footprint of the software, and adding 256 bytes is not what I had in mind. There will have to be a rewrite of how data i stored before merging this into the master branch.

gskjold, 2021-04-07:
Oh I see now that this is actually how Azure IOT works, never mind the question :)

ofredstie, 2021-04-07:
Changed to 1024 and still same result.

gskjold, 2021-09-23:
Committed support for long username and password

closed 2021-09-23 by gskjold