UtilitechAS/amsreader-firmware #109: Unit starts with debugging enabled

ArnieO, 2021-06-25:
Hardware: Pow-U 
Hardware setting:
[image]

After restart, the debugging is active:
[image]

I assume debugging should not be active by default; could cause conflict on units using UART0 Rx pin as HAN-input.

gskjold, 2021-09-23:
Are you sure this is not a residual config from earlier? I am unable to reproduce on a blank board.

ArnieO, 2021-09-23:
It is systematic. For each new blank board I program those two are set when I enter the debug menu.

gskjold, 2021-09-23:
Reproduced with brand new board

closed 2021-09-23 by gskjold