UtilitechAS/amsreader-firmware #115: Kamstrup meter outputting nothing despite Kamstrup verifying HAN is enabled

mortenmoulder, 2021-09-21:
Hi Gunnar and the whole AMS, HAN, DLMS, etc. community

I hope it's okay if I use this issue as "notes", that should show up if people search for the same kind of issues I've been experiencing. Since I have not yet gotten any data from my Kamstrup Omnipower meter, I hope I can keep this issue open until I either get data or give up.

So, I live in Denmark and I have a Kamstrup Omnipower meter. I've had two meters in total. The first meter was old, yet still an Omnipower, and could not receive any newer firmware updates. I was told by Kamstrup that HAN was enabled, but my electricity supplier did not have the HAN tab in VisionAir (a software they use to control meters).

I tried everything including hooking up a borrowed oscilloscope as well as my own logic analyzer. Nothing was showing on DATA_OUT.

After 30+ mails back and forth with my electricity supplier, we decided to upgrade it to a new meter. A week or so later, I received my decryption keys, and was told that HAN was enabled. I was so excited, so I spend the WHOLE evening debugging everything I possibly could. I tried a handful of ESP8266 and ESP32 boards, older firmwares, newer firmwares, beta firmwares sent by Gunnar via mail, hooking up my logic analyzer... everything. Nothing worked.

Today, the 21th of September, I called up Kamstrup in sheer desperation. **I wanted help**. **I needed help**. I'm basically on my knees at this point. And the result? "**We can confirm the HAN interface is enabled, and was enabled the 17th of this month**". Great.

I'm scheduled for a call tomorrow the 22nd with a technical supporter from Kamstrup, whom I spoke with previously, who told me that:

1. Making a device yourself (like using an ESP8266/ESP32 and breadboard wires) is a NO GO
2. They do not support smart-me or any other devices, except the OFFICIAL modules

The few times I've spoken with Kamstrup, they've asked me which HAN device I'm trying to hook up. So far it seems like they don't care AT ALL that DATA_OUT is completely dead.. and has been dead on both my meters. So I keep telling them "My module uses the same hardware as the smart-me", because it essentially does.

Here's a video of me showing that my logic analyzer works and that nothing is being sent by DATA_OUT (even though I don't show a full 10+ second stream of nothing, so you just have to trust me on that one): https://mega.nz/file/ncBCwDwD#PjuUiytbRw4oL_y0grRQRmrxsUNmPJfWlNz_q-VUR6Y

I'm looking forward to the phone call tomorrow, and I will keep you guys updated. If you've gotten this far, send a prayer my way. Please. Thanks for reading.

Also, if you have any clues, feel free to share them. @ArnieO sent me [a link from a Norwegian forum](https://www.hjemmeautomasjon.no/forums/topic/5894-kamstrup-ams-m%C3%A5ler-wifi-adapter/?page=3&tab=comments#comment-91541), where someone got their meter to work.. by getting it replaced.. which I've already done. What are the odds of having TWO faulty meters? Slim to none. You be the judge. Cheers.

ArnieO, 2021-09-22:
All you need to verify whether the meter gives data on the HAN port is a multimeter.
All need to say to your grid company and Kamstrup is that there is no data on the appropriate pin in the meter socket. 
Then ask they to fix it.

I don't think you need to discuss with Kamstrup what you hook up to it, except the measurement device to verify that it outputs data.

mortenmoulder, 2021-09-22:
@ArnieO Thanks for the reply. I do agree that Kamstrup doesn't need to know what I'm hooking up (specifically, at least). However, it seems they are very, very interested in that information, for some reason.

I did hook up a multimeter to GND and DATA_OUT, which showed a steady 3.3v signal with no changes at all. Not even a tiny dip or anything.

ArnieO, 2021-09-22:
I guess Kamstrup is just trying to assess whether anything you did could be the reason for the unit failing - which I find understandable seen from their point of view.

If there is no data frame from the meter, there is nothing you can do except requesting assistance from your grid company.

mortenmoulder, 2021-09-22:
[image]

Well, well, well, would you look at this! I spoke with said Kamstrup employee for about half an hour. We troubleshooted a lot of stuff, but he kept talking about which module I was using. He knew about the POW-K, so I just told him "It's the same hardware as that, and it even runs the same software", so he was definitely on the right track.

After about 5 minutes of pausing, he told me "Go try it again", and low and behold, it freaking worked!!

I asked what was wrong, and he told me something along the lines of "It didn't save properly the first time, so I unticked the checkbox, saved, checked it again, and saved". This was definitely something my own supplier could have done, instead of Kamstrup, but I think we all learned something from this. If anyone has this problem again in the future, that might be the issue.

I was also able to read the data using my logic analyzer and Saleae Logic about every 10 seconds.

Thank you very much for the help, @ArnieO - it's great to talk to someone with some real experience, who can also test and debug at the same time.

ArnieO, 2021-09-22:
Thank you for the feedback, great to hear that you got it up and running!
And Pow-K is my product, so yes - I know it quite well. ;-)

Your experience in this thread is valuable in case anyone else runs into the same issue.

ArnieO, 2021-09-22:
PS: The internal ADC of the ESP8266 is not very accurate, so this "red flag" on the ESP voltage is probably not an issue. 
[image]
The ESP button goes red if the voltage is low, nominal voltage is 3,3V, and that is the expected value right below. 
If you can measure the actual voltage (easiest access is on the leads of the large blue capacitor), you can adjust the ADC by:
System/GPIO:
And adjust the Vcc multiplier until the displayed value is the same that you measure:
[image]
Don't touch anything else on this screen.

mortenmoulder, 2021-09-22:
Definitely - I hope my frustrations will help someone else some day haha.

I asked him, if he knew any other modules "such as the smart-me", and he told me "Well I know a Norwegian guy made some called POW-K for Kamstrup meters", so I was happy to tell him, that I was in contact with both you (the creator) and Gunnar, who both suspected that HAN wasn't enabled, which it ultimatively was not.

mortenmoulder, 2021-09-22:
Is the ESP voltage used for anything besides looks, or is it a visual cue to determine if the board is getting enough current to keep itself up 24/7 using the 4.15v V_IN on the meter?

gskjold, 2021-09-22:
Good to hear that it finally worked out!

Voltage is only for display unless you configure a startup voltage limit in the config, which is only needed for MBUS powered devices and not applicable for a kamstrup module.

closed 2021-09-22 by gskjold

tobvil, 2023-03-06:
Hi @mortenmoulder,

I have the same issues that my net provider doesnt have the license for the VisionAir, so they can't enable HAN Push.

Is it possible to get Kamstrup to enable it remotely? Did you just call their main number, how did you get in contact with them?

mortenmoulder, 2023-03-06:
@tobvil I called up Kamstrup and we managed to fix it, yeah. However, it stopped working a few months back, so now I went to GPK11 key instead with a Watts Live device. To get the data into Home Assistant and so on, I have an IR device called HA Glow.