UtilitechAS/amsreader-firmware #116: Static IP bricks the device

SeMdhl, 2021-09-22:
Worked fine until i set static IP, Stoped working after 2 days (DHCP lease time?) and now i cant set it back to DHCP. Tried moving it to a place without wifi signal, shorting 3.3V and GPIO 13, and SW1 to reset it back to AP mode. Nothing worked so its completely bricked.

Checked the google auto-fill database to see if i messed up the static IP settings. But nothing wrong there, so i guess static IP just does not work.

ArnieO, 2021-09-22:
Static IP works fine, so there must be something else going on. 
Please provide more information on your hardware and version AmsToMqttBridge.

SeMdhl, 2021-09-22:
> Static IP works fine, so there must be something else going on.
> Please provide more information on your hardware and version AmsToMqttBridge.

Could be something else yes, but if i can get it back to AP mode i can just set it up with DHCP again.
Im not the one who set these up, so i dont know the software specifics. But its The AmsToMqttBride V1. I can try to find out the software version, but i doubt they remember.

ArnieO, 2021-09-22:
Which hardware are you using? 
You set it to AP mode by pulling GPIO0 to GND for around 5-10 seconds. 
Depending on your hardware an LED should then start flashing periodically. 
You will then find the access point on your wifi, and you connect to it on IP http://192.168.4.1

SeMdhl, 2021-09-22:
@ArnieO 

Im not sure, The only significant info on the Board is: E2AL3B-ESP-F

I tried pulling GPIO0 to GND, after holding it for 10 seconds it blinks. And then continues to periodically blink, but there is no AP

[image]
[image]

.

ArnieO, 2021-09-22:
OK... this is history time!

This is one of the first boards I know of designed for this purpose, designed by @roarfred.
All documentation is here: https://github.com/roarfred/AmsToMqttBridge

Board layout and schematic: https://github.com/roarfred/AmsToMqttBridge/tree/master/Electrical/HAN_ESP_TSS721

@roarfred was the one who started the development of the firmware, but he has sadly passed away, and @gskjold now maintains it after having updated it significantly.

I had one of those boards running for quite some time until I decided to get rid of the external powersupply, and designed my own boards (see [amsleser.no](https://amsleser.no/)).

The firmware on your board is probably quite old, and I do not remember if it went to AP mode by pressing SW2 (GPIO0). If you have the tooling for reprogramming the ESP to latest version firmware, that might be a good option for you.

I still have that card lying around here somewhere...

SeMdhl, 2021-09-22:
Oh wow, well then it makes sense that it doesnt work with static IP. Since its the first hardware, and probably one of the first softwares. 
SW2 dont seem to do much, and SW1 makes the light stop flickering when i hold it down. If i hold both, then i have to press SW1 afterwards for it to start flickering again. But none of the buttons or combinations makes it go to AP mode. 
The guy that set these up said earlier that it should be a solid blue light when it enters AP mode, which i have never seen happen yet. (Have also checked for it on nearby wifis after every try)

Which ESP tooling are you talking about? I thought the only way to reprogram it was through wifi.

ArnieO, 2021-09-22:
I think you´re right that AP mode was indicated by a steady blue light on the ESP LED on that early firmware. I still think it should go there by holding SW2 down for some seconds - but I might be wrong. 
SW1 is the reset for the ESP. If you hold SW2 then press SW1 the ESP will go to programming mode, and expects program to be uploaded via the 6-pin header contact on the board. If you press SW1 again it exits programming mode and restarts the ESP.
I do not think that old firmware supported wireless programming. Must be done using a USB cable with an FTDI module:
https://www.kjell.com/no/produkter/elektro-og-verktoy/arduino/moduler/luxorparts-usb-til-seriell-adapter-for-arduino-p88064

ArnieO, 2021-09-22:
If you want to dig into the history, this is the thread (83 pages!!) where the design and firmware was born:
https://www.hjemmeautomasjon.no/forums/topic/1982-lesing-av-ams-data-amshan-iot/

SeMdhl, 2021-09-22:
The combinations seems to be right, since if i make it go to programming mode, the light stops flickering. Then if i press SW1, it starts flickering again every 10-15 seconds like it usualy does. 
It really just seems like its not possible to reset it to AP mode. 

But if i find (or buy a new) tool and then flash the program again, would it go to AP mode? If yes, i think thats the only option. Also i could upload a newer version so its actualy possible to reset it back to AP again later.

I dont think they want to upgrade to the newer cards, since this is just temporary. But ill make sure to get the newer boards for the next installation. Not sure what brand the AMS meter is where these are used.

Verry interesting that this whole project started with one thread in a forum.

ArnieO, 2021-09-22:
Going to AP mode is controlled by the code in the firmware, but your firmware is so old (and there were several updates done while @roarfred managed the development) that I cannot say which version you have.

It is important to switch the FTDI module to 3,3V mode, or you risk killing the ESP. Then follow instructions in the Wiki here ("Flashing"), after setting the device in programming mode as explained above.

SeMdhl, 2021-09-23:
Well i think ill just get the latest version, if that works.
Will one of [these](https://github.com/gskjold/AmsToMqttBridge/releases) work for these old boards? Or do i need the [version](https://github.com/roarfred/AmsToMqttBridge/tree/master/Code/Arduino/AmsToMqttBridge) from @roarfred ?

gskjold, 2021-09-23:
Yes, all versions from this fork will work. Download the esp8266 file from latest version

SeMdhl, 2021-09-23:
Allright flashed with the [latest version](https://github.com/gskjold/AmsToMqttBridge/releases), but it still wont go in AP mode. After flashing it still looks like its in programming mode, so i tried SW1, SW2 and cold resetting it. SW2 does nothing, and the other two options just make the light flicker every 20 seconds again but nothing else.

gskjold, 2021-09-23:
Holding SW2 for 5 seconds should put it in AP mode

SeMdhl, 2021-09-23:
Does not work

gskjold, 2021-09-23:
Weird. Lets wipe the chip and then re-flash it with new firmware. When you run through setup, be sure to select hardware "Custom hardware by Roar Fredriksen"

esptool.py --port COM29 -b 115200 write_flash 0x000000 blank_1MB.bin 0x100000 blank_1MB.bin 0x200000 blank_1MB.bin 0x300000 blank_1MB.bin

[blank_1MB.zip](https://github.com/gskjold/AmsToMqttBridge/files/7215359/blank_1MB.zip)

SeMdhl, 2021-09-23:
Yes, finaly it works now. Tested on two of them and it looks like this was the solution for both of them. Also i tested static IP with the new firmware, and that also works now. So i guess it was just the old firmware that bricked them.

Thank you for all the support @gskjold and @ArnieO

gskjold, 2021-09-23:
Excellent news!

closed 2021-09-23 by gskjold

ArnieO, 2021-09-23:
Great news - good to hear!