UtilitechAS/amsreader-firmware #120: Suggestion: Remove "Temperature" and "Energy Price" menus for devices that cannot display them

ArnieO, 2021-09-22:
If not too complicated to implement, these two menu items could be omitted:
- Omit "Temperature" on devices that do not have temp sensor 
- Omit "Energy Price" on ESP8266-based devices

[image]
[image]

gskjold, 2021-09-23:
This is actually a bit harder than one would believe, mostly because the most likely solutions would waste either runtime memory or flash space, which are both stretched in this project for different reasons.

ArnieO, 2021-09-23:
Telnet log indicates that the unit spends around 750 ms trying to read temp on a unit with no temp sensor. Not knowing those details of the code, I imagined it could be possible to «if» around it by testing the setting on the GPIO screen. But if simple solution impossible, we live well with «as is»!

gskjold, 2021-09-23:
It already check if GPIO is set, so it will not query sensors if no GPIO is set.

The "hard part" would be to remove the menu items, as they are hard coded for memory optimization.

gskjold, 2021-11-18:
Found a suitable solution for this, added for next major version

closed 2021-11-18 by gskjold