UtilitechAS/amsreader-firmware #123: Request: Option to ignore List 1 frames sent by Kaifa and Aidon each 2.5 sec

ArnieO, 2021-10-01:
### Background
Some board designs that follow the Norwegian HAN/MBUS standard (RJ45 connector) are powered from the HAN-bus, where there is relatively little power available. This is the case for the Pow-U modules.

There are 3 data lists sent from the meters:
- List 1: Sent by Kaifa and Aidon meters each 2.5 sec
- List 2: Sent by all meters each 10 sec
- List 3: Sent by all meters each hour

Kamstrup got a waiver from the standard, and only sends List 2 and List 3. 

- List 1 contains only Active power imported (kW in from the grid). 
- List 2 contains List 1 plus List version identifier, Active power export (kW out to grid from e.g. your solar panels), Reactive power import and export (kVAr), Meter ID, Meter type, Current per phase, Voltage per phase. 
- List 3 contains List 2 plus Meter clock/date, Cumulative active energy import and export (kWh), Cumulative reactive energy import and export (kVAr).

In order to get sufficient power from the HAN bus, the ESP8266 in the Pow-U module needs to go to automatic light sleep between each Wifi transmission. However, there is a known but not well-documented bug in ESP8266 that makes difficulties.
See for instance [this blog post](https://blog.voneicken.com/2018/lp-wifi-esp-comparison/).

### Issue sought remedied
One consequence of this seems to be that it takes longer than expected for it to go to sleep after wifi activity, and this is especially difficult if it is connected to a Wifi SSID with multicast/broadcast "noise". Apple Bonjour is known to use multicast, for instance in networks containing AppleTV or other units using this protocol. Networks with Chromecast etc may have similar issues.

One user has reported a continuous power-down issue on a Pow-U board. As soon as the Pow-U was connected to another SSID (his guest SSID), the unit worked correctly. 

However, this is not necessarily feasible for all users; some wifi routers may not have the option to set up a separate guest SSID.

### Request
I propose to make an option that can be selected from the GPIO-menu, titled for instance "**_Ignore List1 frames_**".
When this option is selected, the card will not react to List 1 frames (not transmit the data). This could potentially give the ESP8266 enough "peace and quiet" to go to sleep and keep the voltage stable.

The consequence of setting this option will be that Active power import readings from Kaifa and Aidon will only be updated each 10 seconds, even if the meter reports it each 2,5 seconds.

### Testing before release
The option should be tested on a network where the described issue is manifested; my Pow-U user has volunteered.

gskjold, 2021-10-03:
Attached is a firmware file where list 1 is ignored

[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7274299/firmware.zip)

ArnieO, 2021-10-03:
Unfortunately, that did not fix the issue. It stayed live a bit longer, but still goes down.

closed 2021-10-10 by ArnieO