UtilitechAS/amsreader-firmware #126: Request: Support for Danish meters (Omnipower) using GPK11 key

heinekmadsen, 2021-10-06:
Hi...

I'm not sure this is even possible, as i have no deeper knowledge into this topic.
The case is, that i bought a Pow-K (Kamstrup) module from amsleser.no.
I have fittet it into my Kamstrup OmniPower meter and have supplied the GPK60 and GPK61 keys, but i get no data.

According to my power provider they state that it works with the Smart-Me modules using GPK11 key.
the provider does not have access to that part of the Kamstrup software where they can enable that feature.

So my request is, can this software use a GPK11 key to get data?

ArnieO, 2021-10-06:
I propose we first establish whether the meter provides data at all, as there has been cases of meters not having the HAN port open despite the grid company insistance. Does the LED show a blue light flickering for around 1 second every 10 seconds?

ArnieO, 2021-10-07:
If there are no blue flashes there is no incoming data. The red flashes indicate the same as the red HAN status. There is a similar case in one of the threads here, the problem turned out to be that the port was not activated after all.

ArnieO, 2021-10-07:
Here: https://github.com/gskjold/AmsToMqttBridge/issues/115

heinekmadsen, 2021-10-08:
If using a multimeter to test if my meter is sending data, then what should i expect to read?
I assume i need to measure between pin 5 and 6.

heinekmadsen, 2021-10-08:
So...
I hooked up some jump wires to the pins and my multimeter. Used my phone to record a video of the multimeter for 35 seconds.

Result: a steady 3,29v for the whole 35 seconds. Not a single deviation at all.

I assume this means no data is being sent from the meter. And if I used a smart-me module it wouldn't work either, correct?

According to the grid company my meter should be ready for a smart-me module, but it doesn't makes sense to me at all.

ArnieO, 2021-10-09:
So - no data. You seem to have the same situation as the guy in https://github.com/gskjold/AmsToMqttBridge/issues/115

I would contact the grid company and ask them to double check that the HAN port is acivated.

ArnieO, 2021-10-14:
Well, this was surprising. There are around twenty Pow-K boards running in Denmark now, and I have not seen this problem before.

The only idea I can think of is that the meter needs an "enable" signal on an input pin to ensure the meter outputs data. This is not specified for the Kamstrup meters, but is used in the IEC 62056-21 interface standard used in for instance Sweden and Netherlands ([see my overview here](https://github.com/ArnieO/SmartMeterDocumentation)). Those interfaces have a "Request" input that must be set high from the device in order to get data from the meter.

The Kamstrup meter has one input pin, see [this document](https://github.com/gskjold/AmsToMqttBridge/blob/master/doc/5512-2584_GB_A2_11-2019.OMNIPOWER.HAN.interface.specification.pdf), which is pin 2 (figure 5):
[image]

The document states (page 9) that pin 2 is pulled high inside the meter by an 82k resistor. It also states that the pin is used for "serial communication of data to the meter", which is in most countries not used (disabled in the meter firmware, I assume) - to eliminate the possibility of the user being able to do whatever could be done from this port. This is why I chose to have pin 2 on Pow-K not connected.

Now I can only speculate:

- I assume you are 100% sure that you measured on the correct pins a stable (approx) 3,3V.
- **IF** the Kamstrup meter needs the input pin to be high to enable HAN port data **AND** there is an error on that input on your meter (pullup resistor missing or wrong value?) **AND** the creators of the Smart-me decided to connect this pin to a high signal to be on the safe side **THEN** pulling pin 2 high on the Pow-K board could resolve the issue, causing the meter to deliver data.

However, pin 2 is specified for maximum voltage of 3,3V, so you cannot connect it to pin 1.

This is what you could do **if you're comfortable with trying this** and have access to a suitable resistor:

Connect a resistor (I would try 10k, but anything smaller than than 82k should work) between connector pin 2 and 3,3V on the Pow-K board. If you're equipped to solder it in that is fine, just be careful not to short circuit the connector pins, and also do not apply too much heat close to the large capacitor!
Like this:
[image]
If you are not equipped for soldering, try to twist the leads of the resistor in place, and be sure that there is no short circuit to other pins of the connector. 

(Note that there is an unfortunate error on the silkscreen of the card: Vcc and GND on the large capacitor are switched.)

heinekmadsen, 2021-10-14:
Still no blue flashing, only red flashes and red status in the hi

ArnieO, 2021-10-14:
What makes this strange is that what you reported earlier (multimeter measurement) indicates that there is no data coming from the meter.  However, the Smart-me unit works as expected and reads data from the meter.

The blue pin on the LED is directly connected to the incoming data signal, so it will flash providing there is voltage on the board. Which there clearly is - as the ESP8266 is up and running, generates the UI and generates red flashes on the LED.

So let us assume the board **does** receive data from the meter, and look for potential errors that could prevent the blue LED from flashing. Could you be so kind as to check two points:

1. The pin distance on the LED (D1) is very small, so I take great care to verify after soldering that there is no short circuit between pins. Can you check with ohmmeter the resistance between pin 3 and 4 on the LED? And if you have a good magnifying glass, check if you see any indication of a short circuit between them? The blue pin is pin 4, green is pin 3. If there is a short circuit between those, the signal to the green LED would interfere with the data signal. See below indications of D1 pins 3 and 4. 
2. This should not prevent the blue LED from flashing, but is still a point to check: I did have a bad batch of voltage regulators (U1) in the period I made that board - but I did not think I had shipped any board with bad U1. I can see you make good photos: Can you send me a photo showing the marking on U1? The "bad apples" are easy to identify from the marking.

[image]
[image]

gskjold, 2021-10-15:
Regarding the keys: https://github.com/gskjold/AmsToMqttBridge/issues/73#issuecomment-625050664

Just thinking out loud here, could it be that the smart-me module asks the meter for data rather than passively receiving?

heinekmadsen, 2021-10-15:
Well.. that might be a dead-end :(
https://www.ihc-user.dk/forum/forums/topic/7467-kamstrup-elm%C3%A5lere/?do=findComment&comment=58737

A KMP protocol... Will try to reach out to Kamstrup and see if they are willing to hand out some documentation for this protocol.

heinekmadsen, 2021-10-15:
Maybe some handy info here... even though it's rater old: https://github.com/bsdphk/PyKamstrup

gskjold, 2021-10-15:
Thanks, looks similar to the arduino project I referenced ealier. This method only worked before they added encryption.

heinekmadsen, 2021-10-15:
@ArnieO It's called "Nord Energi" https://www.nordenergi.dk/

I'll see if i can find any regulatory info.

ArnieO, 2021-10-15:
I am collecting information on the meters here: https://github.com/ArnieO/SmartMeterDocumentation
You can see there what I have found for Norway and Sweden in this table: https://github.com/ArnieO/SmartMeterDocumentation#national-technical-requirements

If you find regulatory information for Denmark, it would be great if you could report it as an "issue" on that repo: https://github.com/ArnieO/SmartMeterDocumentation/issues

heinekmadsen, 2021-10-15:
I reached out to the grid company once again... They are now willing to swap my meter for a new one to see it that makes it work.

But still think it would be nice to find some knowledge on the encryption on the KMP protocol. Will continue my hunt on this.

heinekmadsen, 2021-10-15:
So the meter guy showed up and we had a long chat about the issues.

We actually ended up not changing the meter, since the info from Kamstrup stated that I wouldn't work changing the meter as the grid company would still need to buy access to that module in Vision Air. So to keep (if any left) good will from the grid company we decided not to change it.

Mausy5043, 2021-10-16:
> There is supposedly some information on Kamstrup Meter Protocol here: [bsdphk/PyKamstrup#3](https://github.com/bsdphk/PyKamstrup/issues/3) I have requested access.

In the linked issue the Google Drive link is proving to be somewhat unreliable. For those interested, I've added the documentation to my own repository on Gitlab: [here](https://gitlab.com/mausy5043-diagnostics/kamstrupd) and updated the issue with the same info.

May I note that most older Kamstrup meters (including the one in my home) here in the Netherlands have an RJ11 socket (called the P1-port) providing a simple serial protocol. I can't imagine Kamstrup would redesign the hardware I/O for each meter. Rather they probably just changed the connector. I'm guessing that the underlying protocol is still a simple serial protocol. The datastream however might be encrypted.   
In my case the data is just inverted. I'm guessing this was done to annoy end-users/consumers :wink: The cable that I use converts the connection from RJ11 to USB and converts the data to "normal". As a result I can use the Python `serial` module to extract the relevant data from the telegrams.

Hope this helps, have fun!

madsbg, 2021-10-16:
> Still no blue flashing, only red flashes and red status in the ui

**TL;DR** I think the HAN port in your meter has not been correctly enabled. If Nord Energi cannot do it, there must be someone who can do it for them.

_All the urls below link to pages in Danish. I hope that it's okay._

I bought a Pow-K module from amsleser.no a couple of month ago, and when I first installed it in the meter, it was flashing red just like yours. The grid company in my area is Radius, and I had requested the encryption key from this page: https://radiuselnet.dk/elkunder/elmaaleren/betjening/ but that is the key needed for the smart-me module.
When I later found out, that I needed two different keys, I called Radius customer service, but the person I talked to wasn't sure exactly what I was asking for, and he said he would forward my request.

A couple of days later I received an email from a company called Nexel (nexel.dk), that apparently are the ones responsible for the backend infrastructure of the power grid and is a contractor for Radius. A nice gentleman asked me if I needed the DLMS/COSEM keys and asked for my meter number and customer number. I presume that he also was the one enabling the HAN port in my meter. The AMS module started flashing blue and after I entered the keys everything has been working great.

What I am getting at is that even if Nord Energi do not have access to enable your HAN port from their system, they must have a contractor running the infrastructure for them that can. Maybe Nord Energi customer service just doesn't know about it, because so few customers ask for it.

I first found out about amsleser.no from this article: https://www.version2.dk/blog/faa-indsigt-dit-elforbrug-1092934. Both the article and the comments contain lots of good information also about the legislation (comments 29, 30 and 31).

> Bekendtgørelse om fjernaflæste elmålere og måling af elektricitet i slutforbruget:
> https://www.retsinformation.dk/eli/lta/2019/75
> 
> § 7. Det skal være muligt for forbrugeren efter åbne standarder at tilkoble eksterne enheder til den fjernaflæste elmåler og løbende udtage forbrugsrelevante data, jf. § 5.
> Stk. 2. Overførsel af data skal være sikret via kryptering eller lignende.
> Stk. 3. Tilkobling af eksterne enheder skal kunne foretages af lægmand uden at bryde plomber eller lignende. Netvirksomheden kan beslutte, at aktivering skal foretages eksternt af netvirksomheden. Dette må ikke påføre forbrugeren unødige ekstraomkostninger.

In short: The power companies in Denmark are required by law to give the customer access to retrieve data from the meter using open standards.

heinekmadsen, 2021-10-17:
@madsbg 
Seems you have a lot of knowledge and access to Vision Air?
Can you confirm that the part of Vision Air where you set the settings for the HAN interface is not available until the grid company has bought access to it?

I'm not aware of any sub contractors under Nord Energi. The meter guy wo was so kind to show up last week was from Nord Energi.

I have tried to reach out to the Home Assistant community in Denmark to find other potential Nord Energi customers with info on this, and asked if they could ask Nord Energi for the HAN-NVE to be enabled and get the GPK60 and 61 keys to put pressure on Nord Energi.

I'll try to pursue the "regulatory" way aswell, as i don't see Smart-Me as an "åben standard" (open standard) with "unødig ekstra omkostninger" (unnecessary extra costs).

madsbg, 2021-10-18:
> But do nexel work for Nord Energi aswell?

No. But I thought that being in the same industry, he might know who to contact.

tha1971, 2021-10-18:
Hi 
I am customer 9800.
I have almost  given up on my Pow-K. I am waiting on an opportunity to replace my old Kampstrup with a new Omnipower.
My symptoms are fairly similarly to Heines.
But is it correct that it may even solve the problem?
Because of a "closed" protocol by Kamstrup and lack of money by Nord Energi?

heinekmadsen, 2021-10-19:
> > But is it correct that it may even solve the problem?
> > Because of a "closed" protocol by Kamstrup and lack of money by Nord Energi?
> 
> Hi @tha1971 - thank you for joining the discussion! It is fully understandable that you have given up on Pow-K. Please feel free to return it to amsleser.no to get a refund.
> 
> I don't know if "lack of money" is the right term, but my understanding is that Nord Energi Net refuses to do a firmware upgrade of their Kamstrup meters due to cost.
> 
> And Yes - the protocol needed to "talk to" the elmeter (to "pull" measurement data) is a protocol that is not public. I have seen it referred to as "proprietary and NDA protected KMP protocol" (KMP = Kamstrup Meter Protocol). [For instance here.](https://www.gurux.fi/comment/13590#comment-13590)
> 
> As far as I can understand, Nord Energy Net then violates §7 in [BEK nr 75 af 25/01/2019](https://github.com/ArnieO/SmartMeterDocumentation/blob/main/Denmark/B20190007505.pdf) that requires open standards: [image]

I sure hope that reading the flashing led from the meter is not an open standard... if that's the case, then it's definitely a dead end.

ArnieO, 2021-10-19:
> I sure hope that reading the flashing led from the meter is not an open standard... if that's the case, then it's definitely a dead end.

That would not satisfy §7 here: "...og løbende udtage forbrugsrelevante data, jf §5."

...where §5 says:
[image]

The flashing LED only gives an indication of current power consumption, nothing else.

henriklund, 2021-10-19:
One important note is the age of your meter:
[image]
Ref. §9, stk. 5 expected lifespan is 15 years.

henriklund, 2021-10-19:
Then I would def. ask Nord Energi what they are doing to comply with §4-7 :)

chmadkk, 2021-10-20:
> Well... mine's from 2015... so that must be on the safe side 😁

Hi Heine. Another dane joining in. 
ive had a little chat with my net provider. Which is Flow/SEF. 
they repliks that they use the “legacy” meter and not the newer omnia. And the legacy does not use push. But only pull..

i have mine Working using the IR port on the meter. Using some Old code from mtrab. But would like a more integrated system.  And if you look at the code. The ir head requests data with an start command. 

Is it possible that the ams port is just An wired version of the ir head ? 

Its well dokumented in the PyKamstrup repo.

heinekmadsen, 2021-10-20:
> Hi Heine. Another dane joining in. ive had a little chat with my net provider. Which is Flow/SEF. they repliks that they use the “legacy” meter and not the newer omnia. And the legacy does not use [...]

Nice to see more Danes joining in...

Well... I know there's other ways of getting meter data, like the Smart-Me module, Frient's zigbee optical thing and alot of diy optical read out from the LED. Currently i'm actually using a Smart-Me module.

The issue in all this is a meter company failing to comply with the regulatory and refusing to buy access to a module from Kamstrup to enable this feature. The regulatory states it need to be an open standard, which the Smart-Me module isn't.

What data are you able to read out from the IR?

chmadkk, 2021-10-20:
Exactly the same data as thru the HAN port. 

Its documented here. 

https://github.com/bsdphk/PyKamstrup

Ive asked my supplier how they are seeing the KMP port as an solution to regulatory where it states that the protocol has to be open and documented.

RasmusAaen, 2021-11-15:
Just another dane with a backward power grid supplier chipping in :-)
I'm in northern Denmark as well (zip: 9970). I'm currently have a conversation with a guy from Elinor - the grid owner for a small section of the map presented earlier:
[image]

I initially requested the GPK60/GPK61 keys to which they replied that the Omnimeter at my house could not be used to read usage data. After asking them how they conformed to §7 of the law mentioned above, they told me that they supported the GPK11 key and smart-me module, and thereby was fully compliant with the law. Sigh...

I have asked then for the GPK11 key and documentation of the open standard protocol - looking forward to hearing the response.

heinekmadsen, 2021-12-08:
@RasmusAaen any news on the GPK11 key documentation?
I have the same request at Nord Energi who is in dialog with Kamstrup for some documentation. (They say.............).
According to Kamstrup, the Meter company does not have to enable HAN-NVE to be compliant with the regulatory.

RasmusAaen, 2021-12-08:
No, they have "sent my questions further up" in the system and will get back to me. This was on Dec. 1.
I questioned the legality of the smart-me module, since it sends all meter data to Switzerland, which is not part of the EU. I believe that is a violation of GDPR rules, or at the very least requires them to have a data exchange agreement with the company behind the smart-me module.

Very much looking forward to hearing from them.

heinekmadsen, 2021-12-08:
Well.. that’s an interesting take on this… maybe the GDPR card is the way to go… Let me hear if you get any info from the company.

heinekmadsen, 2022-01-20:
Great news!!!!!
Received info from Nord Energi today, that they are working with Kamstrup to get the correct module in Vision Air software...

It's been a hard battle, and still not quite there yep, but getting closer :D

RasmusAaen, 2022-01-20:
Fantastic - I might be able to use this to push Elinord to do the same. 
The last I heard from them was two weeks ago where they informed me that the issue had been forwarded to the legal department of their trade association (brancheforening), so I guess they took the GDPR issue seriously :-)

@heinekmadsen - would you be willing to email me the details of who you have been communicating with in Nord Energi - I'll suggest that my Elinord contact could get in touch with Energi Nord to get some info on how they are doing it. My email address is rasmus@aaen.net

ArnieO, 2022-01-20:
Thank you for update - and would appreciate new update either here or on email to post@amsleser.no when this is resolved.

hcaulfield, 2022-01-27:
Hello, 

I am also trying to read data from the Kamstrup Omnipower. I have used an optical probe and the Gurux xDLMSDirector to access the meter data. However it provides only general data, i.e. meter serial number and several other items.

To get the energy register data, I have tried the python script which uses the KMP protocol, again using the optical probe over a serial interface. 

While I can connect, the energy register data cant be read. So either remote reading of data in this way is not activated, or I need the infamous GPK key or keys. 

Is there any public domain information on the GPK key? Is it again a proprietary key format, or does it follow some standard?

I think I've read that GPK stands for General Public Key.  Is this correct or is it a Group Public Key?

Any info appreciated,
Thanks

RasmusAaen, 2022-02-02:
Update! I received a reply from Elinord today stating that they have made an agreement with Kamstrup to start implementing the solution with the GPK60/61 keys. They wouldn't give me a specific date as the rollout was still being planned, but hopefully it will be within the next few months.
I'll write an update when I hear from them again.

blixtdevelopment, 2022-02-02:
That is good to hear, at least some companies are starting to play nice. Netselskabet N1 is still not confronting with my request and states that they are complying with the law by only giving me their GPK 11 key, which they state is also for an open standard.

ArnieO, 2022-02-02:
@RasmusAaen 
Thank you for the update, this is good news!

@blixtdevelopment 
Then you should insist that they provide you with documentation for that "open standard".

blixtdevelopment, 2022-02-03:
@ArnieO 
Trust me I am. Hoping some other N1 customers will chip in and ask for the GPK60/61 keys and/or documentation for how to use their _open standard_.

ArnieO, 2022-02-08:
New information, maybe useful to some of you in Denmark!

It seems that with Radius (large grid company on Sjelland) you can order two types of encryption keys:

- **One-key** (GPK11, I think) is for Pull protocol. If you order this, they activate data Pull (which relies on proprietary Kamstrup protocol)
- **Two-key** (GPK60/61 consisting of Encryption key and Authentication key). If you order this, they activate data Push - exploitable without knowing the Kamstrup protocol.

It would be interesting if @blixtdevelopment and/or @RasmusAaen could contact their grid company and request GPK60/61 keys - at least to see how they respond.

blixtdevelopment, 2022-02-09:
@ArnieO I initial ordered GPK60/61 from N1. They replied that they are not able to share them with me (refuse to say why). They could only give me the GPK11 key and refer to the SmartMe module. They also state that I should be able to measure pulses between pin 5 (DATA_OUT) and 3 (NC). I am not able to, and all the documentation I have found states that pin 4 (NC) and 3 (NC) is not in use - so that seems weird. I speculate that it is because they have not "bought" that ability from Kamstrup, but again, that is just my guess. 

He also states that the HAN module cannot be used with N1's metes, only the SmartMe module can be used to measure usage etc. In my mind that seems strange since the SmartMe module uses the HAN module to communicate, correct? Maybe I'm missing the interfaces and communication protocols. 

Anyways, as for I'm concerned, they are still not compiling to the danish law. I will continue my request and in the end, I will submit the case to [forbrugerombudsmanden ](https://www.forbrugerombudsmanden.dk/) which I guess is the correct danish authority to handle these type of cases.

ArnieO, 2022-02-09:
> @ArnieO I initial ordered GPK60/61 from N1. They replied that they are not able to share them with me (refuse to say why). They could only give me the GPK11 key and refer to the SmartMe module. They [...]

My understanding is that meters using "Pull protocol" (with GPK11 key) use pin 2 (DATA_IN) to request data from the meter. See [my overview](https://github.com/ArnieO/SmartMeterDocumentation#han-interface-description) (not yet updated according to my previous posting here). It sounds strange indeed that you could measure anything against pin 3 (NC/Not Connected according to Kamstrup documentation). So my best guess is that was an error.

The issue here is the firmware in the Kamstrup meter, that can be upgraded wirelessly through the backhaul radio communication network that the grid company use to read data from the meters. Rumors are that Kamstrup is known for their "everything has a cost" policy, so I am quite confident your grid company resists upgrading the firmware in their Kamstrup meters due to a high price tag. 

There are more than one HAN module:

- HAN module with RJ45 connector (also called "HAN-NVE module", as it was initially made to comply with Norwegian NVE standard):  https://www.utomhusliv.se/produkt/kamstrup-han-modul/
The output signal from this module is an M-bus signal (24/12V)
[image]
- HAN module with RJ12 connector, to comply with a newer interface specification used in Sweden: https://www.utomhusliv.se/produkt/kamstrup-han-modul-p1-rj12-anslutning/
The output from this module should according to the Swedish standard be as here below. I have not been in position to get this verified on a Kamstrup meter with this HAN module, but have a test board running in Sweden on an other meter brand with this interface.
[image]

I have the **HAN/RJ45 module**, and can confirm by visual inspection of the board that pin 2 is not used (NC). So it seems evident that meters that require data pull using pin 2 cannot use this module. 

The **HAN/RJ12 module** has a signal named "Request", but when tied to 5V it will cause the meter to push data at regular intervals (verified on a Swedish meter, as mentioned). So it is safe to assume that this module is unlikely to work with a "Pull" meter requiring injecting a proprietary command on pin 2 (DATA_IN) on the 6-pin connector.

Based on this, the information you received seems correct: A Kamstrup meter set up for data pull cannot use any of those HAN modules.

As long as a proprietary Kamstrup protocol is needed to get data from the meter, it seems clear to me (but I am not a lawyer, just a humble engineer...) that those meters do not comply to applicable regulations. 

**And the saga continues - in Switzerland!**
I just got information yesterday evening that there is a similar situation in Switzerland, where a grid company is rolling out encrypted Kamstrup meters with a single encryption key - and advertising the SmartMe module. So we assume it is "data pull". It seems that Swiss regulations state:
[image]
...which Google translates to: _upon request, the grid operator must inform the end consumer, producer or storage operator of the technical specifications of the interfaces of his electricity meter._
So the situation is quite similar to some Danish grid companies. 

@gskjold : This thread is becoming generally interesting, and seems to affect more than one country. I propose changing the thread title to: **Request: Support for Kamstrup Omnipower meters using GPK11 encryption key**

RasmusAaen, 2022-02-09:
@ArnieO - My grid company initially provided me with the GPK11 key, but only with reference to the smart-me module. They would not provide any information on the protocol or HAN port configuration.
Using a multimeter, I don't see any fluctuations when measuring voltage between pin 5 and 6 on the HAN interface, so I believe that my meter is currently configured in pull-mode compatible with the smart-me module.
As I wrote earlier, they have since committed to providing the GPK60/61 keys when they get the meter firmware updated. I'll post an update when I hear back from them.

hcaulfield, 2022-02-09:
Given the security difficulties with getting energy readings from Kamstrup meters, does anyone have any experience of using a split core current transformer and non invasive load monitoring techniques.  

For example [https://www.smappee.com/](url)

ArnieO, 2022-02-09:
> Given the security difficulties with getting energy readings from Kamstrup meters, does anyone have any experience of using a split core current transformer and non invasive load monitoring techniques.
> 
> For example [https://www.smappee.com/](url)

The question is quite a bit out of context for both this thread and this firmware in general. 
Which "security difficulties"?

hcaulfield, 2022-02-09:
The manufacturers seem reluctant to share these GPK keys. These are the security difficulties.

ArnieO, 2022-02-09:
> The manufacturers seem reluctant to share these GPK keys. These are the security difficulties.

I sincerely don't understand what your issue is.
Smart meter manufacturers do not manage the encryption keys; that is done by the meter owners (grid companies). I have never heard of anyone having difficulties obtaining encryption keys from their grid company.

Again, I see no relevance to the subject in this thread.

heinekmadsen, 2022-02-09:
@hcaulfield this is actually a feature request to support the gpk11 key and Kamstrup protocol.
All the companies are willing to hand over the gpk11 keys.

Some companies also share the gpk60/61 keys, but they won't work since the company hasn't bought access to the part of the Kamstrup software which enable the use of gpk60/61 keys.

When a company refuse to hand over the gpk60/61 keys I believe it's either a matter of them knowing that the meters won't support it, or the company is just not well enough educated in the meters and the software.

hcaulfield, 2022-02-09:
Thanks for the explanation.

ArnieO, 2022-02-10:
@jayquegmailcom : See https://github.com/gskjold/AmsToMqttBridge/issues/216#issuecomment-1030770141
There seems to be a "HAN-push aktiveret" tickbox in the tool that the operator will use to activate the HAN port and configure your meter.
Current hypothesis is that (at least in Denmark) ticking this box releases GPK60/61 keys.

blixtdevelopment, 2022-02-10:
I am trying to map all of these GPK keys etc. to all of the danish grid companies. In order for us as users to have an overview but also, hopefully, to pressure other companies to comply with the danish regulations. 

Feel free to add any information that you have [here!](https://docs.google.com/spreadsheets/d/16Pa1HgYTLvXbhNzzlKoMawXdFuOnqigUZ_-ktVJaLcU/edit?usp=sharing) (it is in danish though but most of it should be self explanatory).

henriklund, 2022-02-10:
Just updated it with the order site for Radius.

jayquegmailcom, 2022-02-11:
@henriklund Thanks for the note - it was actually pretty easy - Radius Net provider, the webpage request provides you with the GPK11 key, if you write them back and ask for the DLMS/COSEM keys, they will send you these keys.

heinekmadsen, 2022-02-11:
Got confirmation from Nord Energi that they have purchased access to the module in Kamstrup software, and awaiting activation from Kamstrup 👍🏼

ArnieO, 2022-02-11:
> Got confirmation from Nord Energi that they have purchased access to the module in Kamstrup software, and awaiting activation from Kamstrup 👍🏼

This is great news!

libo, 2022-02-15:
I can confirm that Radius in DK releases the GPK60/61 keys on demand.
I wrote them and they kindly did it for me. Awesome!

You need to fill in this form https://radiuselnet.dk/elkunder/kundeservice/kundesupport-formular/
I leave you the text here as a template :)
```
Kære Radius

Jeg har købt et HAN modul til min Kamstrup elmåler, som nu spørger efter et sæt krypteringsnøgler og authentication-nøgler

Jeg skal bruge GPK60 og GPK61 for at kunne snakke med måleren (DLMS/COSEM)

Derudover vil jeg også gerne have slået HAN interfacet til, så jeg kan snakke med mit HAN-NVE “push” modul.

Kan I sende de to nøgler til mig?

På forhånd tak
[your name here]
```

libo, 2022-02-15:
@ArnieO now... it still doesn't seem like working :-( 

How do I ensure that HAN-push is active and that I am receiving messages? 
When I telnet to the device in debug mode the only thing I see is a
```
(D) (loop)(C1) Used 0 ms to update temperature
```

By looking at [the code ](https://github.com/gskjold/AmsToMqttBridge/blob/29524c2123bfb5b98ace5846b7944cd59885d92d/src/AmsToMqttBridge.ino#L731) I should see some `Frame dump ` messages correct?

libo, 2022-02-15:
Now it works.
[image]

[131 more replies]

closed 2024-01-28 by ArnieO