UtilitechAS/amsreader-firmware #139: A lot of noise in the data

fritsduus, 2021-10-31:
I am using a generic ESP8266. I got a problem with a lot of "bad data packages".  Initially I got a good data package through once and a while. Looking at the debug messages using Telnet I got a lot of checksum errors. I tried moving the wires around, switched to another pin for RX and tried with another powersupply, but it just got worse. 
Do you have any idea what I am doing wrong or what my problem is ?

Here is a filtered list of events received in the Telnet prompt:
(D) Header checksum is incorrect E69A
(D) 7E A0 2B 21  13 23 9A E6
(D) Incorrect frame format E0
(D) 7E E2
(D) Source address length incorrect
(D) 7E A0 E2 FE  E7 F8 E8 0A  12 10 00 FF
(D) Incorrect frame format F0
(D) 7E F6
(D) Header checksum is incorrect FE9A
(D) 7E A0 2B 21  13 23 9A FE
(D) Header checksum is incorrect FF00
(D) 7E A0 2B 21  13 E3 00 FF
(D) Header checksum is incorrect 00
(D) 7E A0 E2 E2  E7 00 0F 00  00 00
(D) Header checksum is incorrect C00
(D) 7E A0 BC E3  9A FE FF 00  00 0C
(D) Incorrect frame format 20
(D) 7E 2B
(D) Incorrect frame format E0
(D) 7E E2
(D) Incorrect frame format B0
(D) 7E BC
(D) Incorrect frame format E0
(D) 7E E0
(D) Header checksum is incorrect C00
(D) 7E A0 E2 23  9A DE 0F 00  00 0C
(D) Incorrect frame format F0
(D) 7E FE
(D) Incorrect frame format 20
(D) 7E 2B
(D) Header checksum is incorrect 00
(D) 7E A0 23 FE  E6 E7 FF 00  00 00
(D) Incorrect frame format 30
(D) 7E 31
(D) Header checksum is incorrect 00
(D) 7E A0 BC E3  9A FE FF 00  00 00
(D) Incorrect frame format E0
(D) 7E E2
(D) Header checksum is incorrect FE9A
(D) 7E A0 2B 21  13 23 9A FE
(D) Destination address length incorrect
(D) 7E A0 FB A2  FC E6 FF
(D) Header checksum is incorrect E7FA
(D) 7E A0 E2 21  13 3F FA E7
(D) Header checksum is incorrect E6
(D) 7E A0 BC A5  FF 9A E6 00
(D) Incorrect frame format 40
(D) 7E 4E
(D) Incorrect frame format E0
(D) 7E E2

gskjold, 2021-10-31:
Not easy to tell, but maybe if you provide some information on hardware (ESP board, MBUS board, meter)

First thing that comes to mind is that the MBUS board is powered by the BUS. Some boards need components removed to function properly. Another thing could be that the MBUS runs on 5V and in turn outputs 5V signal instead of 3.3V

ArnieO, 2021-10-31:
@fritsduus I see you are in Denmark, so presumably connected to a Kamstrup device, which outputs 3,3V data signal. 
The power output of the Kamstrup meter is 4,15V. 
The ESP8266 would happily run on that voltage (a bit higher than it prefers, though), but if your input data signal is 3,3V, that could be the issue. The 4.15V should be regulated down to 3.3V using a LDO voltage regulator to power the ESP.

But this is only guesswork until you provide description of your hardware.
https://github.com/ArnieO/SmartMeterDocumentation

fritsduus, 2021-11-01:
Thank you for getting back to me. Yes, I am connecting to a Danish Kamstrup meter with this card: https://minielektro.dk/nodemcu-udviklingskort-med-esp8266-wifi-modul.html, like this:
[image].
The ESP8266 module is powered through the USB connector. I connected GND between the Kampstrup meter and the module and the meter DATA_OUT is currently connected to GPIO14 (I have tried different pins all with similar results).
The AMS reader typically looks like this: 
[image]
When a good package goes through the HAN indicator turns green for at short periode.

ArnieO, 2021-11-01:
OK, thanks. 
Please open System/GPIO and take a screenshot of the settings.

fritsduus, 2021-11-01:
GPIO settings looks like this:
[image]

ArnieO, 2021-11-01:
Thanks, this looks fine. I see from your previous posting that the voltage is on the low side, should be around 3,3V. Have you tried with an other powersupply?

fritsduus, 2021-11-02:
Thank you for taking time helping me out.
I had tried with 2 different powersupplies without any luck, but now trying with a third it looks better. It still says 2,99V, but there must a small important difference.

gskjold, 2021-11-03:
I suspect nodemcu has a similar voltage divider as d1mini and should probably have a 1.1 voltage multiplier in config. But I would check with multimeter to confirm.

Attaching a work in progress firmware you could try. It has a completely different way of reading the UART, so I am curious if that would make any difference to your problem.
[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7465599/firmware.zip)

It may also contain some bugs, but last time I tried it held up just fine on my Aidon meter at least.

mikkel75, 2021-11-03:
Interesting - does the new method support hard UART on GPIO13 (Serial swap)?

gskjold, 2021-11-03:
Not yet, but it is in the pipeline. I've just finished datagram parsing this weekend, so hardware configuration changes have not been touched yet.

ArnieO, 2021-11-03:
> I suspect nodemcu has a similar voltage divider as d1mini and should probably have a 1.1 voltage multiplier in config. But I would check with multimeter to confirm.

Yes it has the same voltage divider, but his setup indicates that he does not use a GPIO to read voltage, so the value should be the one read internally by the microcontroller. That is why I beleive the indicated value is fairly correct - but should be checked by a multimeter measurement.

gskjold, 2021-11-03:
I don't use GPIO on D1 mini either, but need 1.1 multiplier to get correct voltage. Apparently the voltage divider is connected to the ADC. I never understood how that works, but that's what I've read... Seems like the same could be the case for NodeMCU: https://i0.wp.com/www.esp8266learning.com/wp-content/uploads/2016/08/NODEMCU_DEVKIT_SCH.png

ArnieO, 2021-11-03:
Ok, then the voltage is probably OK. I have no hands-on experience with the node-mcu module, but looking at it I do not see much capacitance. And this application reconnects to Wifi often, so voltage stability could be an issue. I would try to add a low ESR capacitor to the 3,3 volt side and see if it helps.

fritsduus, 2021-11-03:
Thank you for all your suggestions. I tried setting the multiplier to 1,1 and that made a  real improvement. Now the ESP indicator is all green and the HAN indicator is green 95% of the time. The overall operation looks much more stable now. I will give you an update tomorrow night when I’ve monitored it some more.

ArnieO, 2021-11-04:
The multiplier only makes a scaling of the ADC value to present the voltage.
So there is another reason for the HAN indicator now being green most of the time. 
Power supply change, probably.

fritsduus, 2021-11-04:
Reading on the HAN interface looks pretty stable now, but I do have problems reaching the website rather often. I guess an ESR capacitor could potentially solve this. Could you recommend a specific capacitor ?

gskjold, 2021-11-04:
Problems loading the web page could mean that the software is busy doing something else. I have a problem guessing what that could be at the moment, but just saying it could be software issue. When the page finally loads, what does it show for uptime?

gskjold, 2021-11-05:
Interesting. And how is the RSSI?

ArnieO, 2021-11-05:
Could also be a network (LAN) issue?

fritsduus, 2021-11-05:
Yes, your right it could be a network issue, but the RSSI is typically -67dBm and rarely changes.

ArnieO, 2021-11-06:
Agreed, RSSI is not the problem, I was more thinking crowded channel or other congestion issues. You can check situation with Wifi analyzer app on your mobile phone. Then, If your router allows, you could try switching your SSID to a less crowded channel.

fritsduus, 2021-11-06:
I live so far out that there are no other wifi networks in reach, so crowded channel is probably not the issue here…

ArnieO, 2021-11-06:
> Could you recommend a specific capacitor ?

The ESP8266 needs good decoupling, especially when wifi is used heavily. 
Any large low-ESR capacitor would do. A tantalum is one option (don’t reverse polarize it!), but you also get quite large ceramic caps. Chips are in general better than leaded.

kng, 2021-12-06:
Hi, thanks a lot for this awesome project!
I am using pretty much the same setup, esp8266 nodemcu v3 and kamstrup non-encrypted swedish. 
The gpio settings for han port I have is UART2 GPIO13 and it works really well. Direct 2 wire connection and no extra caps or such, powered by a power bank at the moment.

[image]

ArnieO, 2021-12-06:
@kng 
Good to hear, Daniel!

If you want to get rid of the power bank: 
On the Kamstrup there is 4,15V power available on the connector where the HAN adapter would normally sit, see https://github.com/ArnieO/SmartMeterDocumentation

I do not recommend using the 4,15V directly; use an LDO voltage regulator down to 3,3V.
Or maybe you could use the voltage regulator on the NodeMCU, if it can handle input voltage as low as 4,15V.

kng, 2021-12-06:
> On the Kamstrup there is 4,15V power available on the connector where the HAN adapter would normally sit, see

Yeah, I have verified this voltage earlier but with the onboard LDO it either brownouts or the supply goes to foldback, So I am planning for this but haven't completed it just yet. I have also seen statements of as low as 75mA available on this port, then it would be much harder to solve.

ArnieO, 2021-12-07:
75mA is according to Kamstrup documentation.

mikkel75, 2021-12-07:
On my module (https://github.com/mikkel75/KamstrupHAN) I have a small DC/DC switcher running only from Kamstrup. I have lowered the ESP voltage to 3 and monitor the battery side of the DC/DC which seems stable over the past month now...

kng, 2021-12-07:
Hi all, I also got it to up and running on a ESP-01, by modifying platformio.ini with: `board = esp01`, `board_build.ldscript = eagle.flash.1m192.ld`
Unfortunately it also draws too much current to be supplied directly by the kamstrup meter via LDO. So I went for the super sketchy solution with a small protected lipo in parallell with the 4.15V, then through a MCP1703. Seems to work just fine :P The average current is ok.
[image]

gskjold, 2021-12-10:
@fritsduus 
Considering other similar issues, try connecting the meter to GPIO 13 and set UART2 in GPIO config. Use newly release v2.0.0

closed 2021-12-13 by gskjold

fritsduus, 2021-12-14:
Sorry for not answering sooner… to much work lately !? I tried upgrading (and changed to GPIO 13). I really like the UI update with the 2 history graphs. I still have problems connecting now and then, but since I seem to be the only one with that problem I guess it must be something with my Wifi setup.