UtilitechAS/amsreader-firmware #140: ESP8266 sometimes HAN 'locks-up' without error in LOG

mikkel75, 2021-11-04:
**Describe the bug**
I see from time to time that the HAN i/f turns red, and no data is transmitted.
After a while (could be hours or seconds) it can turn green again.
This is without showing any error in the telnet log, set to info.

"
(I) (HanReader)Got valid DLMS data (480 bytes)
(I) (HanReader)Decrypting frame
(I) (HanReader)HAN data is valid, listSize: 65
(I) (readHanPort) Using timestamp from meter
(I) (HanReader)Got valid DLMS data (480 bytes)
(I) (HanReader)Decrypting frame
(I) (HanReader)HAN data is valid, listSize: 65
(I) (readHanPort) Using timestamp from meter
(I) (HanReader)Got valid DLMS data (480 bytes)
(I) (HanReader)Decrypting frame
(I) (HanReader)HAN data is valid, listSize: 65
(I) (readHanPort) Using timestamp from meter
(I) (HanReader)Got valid DLMS data (480 bytes)
(I) (HanReader)Decrypting frame
(I) (HanReader)HAN data is valid, listSize: 65
(I) (readHanPort) Using timestamp from meter
  --  HERE THE HAN turns red without futher log ---
"

**Hardware information:**
 - Meter: Kamstrup Omnipower
 - AMS reader: ESP8266

**Relevant firmware information:**
 - Version: 1.5.5 and 1.5.6 tested
 - MQTT: yes
 - HAN GPIO: GPIO13
 - Temperature sensors : No

**Additional context**
I have noticed that I can 'trig' the HAN i/f to work again by going into GPIO settings and press save.
After a while it will however fail again - and again it could take hours or minutes...

Current version doesn't support Serial.swap() and uses SW serial for GPIO13 - wondering if it is a problem related to this...

ArnieO, 2021-11-04:
Can you describe your HW a little better? 
Do you use external power supply, and if so - have you tried with an other one?

mikkel75, 2021-11-04:
Hi,

I use my own HAN module: https://github.com/mikkel75/KamstrupHAN

It is sourced using an external USB dongle - and no I haven't tried another. There is a MPS DC/DC switching the 5V in to 3V to the ESP.

If the problem was related to HAN i/f data being read incorrectly I would expect to see CRC error - which I don't. So the i/f from 4.xxV from Omnipower to 3V ESP should be OK.
I'm thinking something timing related in the SW Serial...

/Mikkel

ArnieO, 2021-11-04:
Interesting - thank you! I suspect your issue could be voltage glitches. 

You can pull 75 mA from the Omnipower, which is more than sufficient. My [Pow-K](https://amsleser.no/hjem/1-pow-k-kamstrup.html) boards use the Omnipower supply and a 1 F supercap *on the 3,3V*. The supercap should be a low-ESR type. The round super cap indicated on your layout usually have relatively high ESR.

With an external power I would still use a low-ESR supercap to ensure the ESP has enough energy available on 3,3V. But it should be on the 3,3V, not on the input voltage side.

If you have an oscilloscope available, check voltage close to the ESP and see what happens when the ESP transmits.

Also consider:

- An LDO linear voltage regulator (e.g. TPS73633) will work fine on this board. If you use the Omnipower supply (4,15V) the voltage drop to 3,3V is small. Average current draw (on memory) with the AmsToMqtt firmware is around 30-35 mA, so power consumption of such an LDO is around 25 mW. From external 5V supply the consumption will be a bit higher: 50 mW (still insignificant). 
- The L on your buck regulator seems to not be a power inductor? That inductor could be limiting the current, as the current to your board passes through it - and be the cause of your issues.
[image]

mikkel75, 2021-11-04:
Hi - and thanks for the input!
Looks great you PCB - didn't know of it... what is the PN of the supercap you use?

Currently I'm running from USB and the supercap isn't mounted - but I will move to that once I get a stable setup.
I will try to add a large cap (~400uF) after the inductor - that should be able to handle the current spikes when transmitting.

Regarding the supercap - I don't see why a low-ESR is needed as it sits before the MPS and has cap's in parallel - have you done measurements to substantiate? :)

Again - thanks for your input!

mikkel75, 2021-11-04:
@ArnieO Just out of curiosity - your Pow-K module, does that have direct connection from Omnipower DOUT to the ESP GPIO? Do you terminate or level-translate the signal?

Thanks.

ArnieO, 2021-11-04:
> what is the PN of the supercap you use?

I buy it from China (Alibaba.com) to get a reasonable price. 
https://www.alibaba.com/product-detail/Combined-Super-capacitor-5-5V-1_62355506797.html?spm=a2756.order-detail-ta-ta-b.0.0.1f602fc2eEu0ZI
You can find them also on Aliexpress, but unless you buy a very small quantity, Alibaba is less expensive, even after shipping cost is added. 
I have seen them also at Mouser and Digi-Key - but the price is *much* higher.

> Regarding the supercap - I don't see why a low-ESR is needed as it sits before the MPS and has cap's in parallel - have you done measurements to substantiate? :)

I agree. Low-ESR is not needed it its current location on your design, that is why pointed out it should be placed on the 3,3V side. But as mentioned, I suspect the inductor you use on your buck converter is a bottleneck. On the [Pow-U](https://amsleser.no/hjem/7-pow-u-universal.html) design intended for the Norwegian M-bus meters I generate 3,3 V from the 24/12V M-bus line using a buck converter. You will find on the rendering the power inductor used. AFIK the chip inductor type you use is intended for RF tuning etc, and has a quite high serial resistance.

> I will try to add a large cap (~400uF) after the inductor - that should be able to handle the current spikes when transmitting.

Good idea, but don't expect an electrolytic cap to do the job, as the current draw spikes are quite violent. You need low ESR.

> @ArnieO Just out of curiosity - your Pow-K module, does that have direct connection from Omnipower DOUT to the ESP GPIO? Do you terminate or level-translate the signal?

Direct connection. The DOUT signal is a 0/3,3V signal, so no level conversion is needed.

mikkel75, 2021-11-04:
The inductor mounted is a NR201610 1uF
[image]

I'll add ceramic caps 2x330uF near the ESP.

ArnieO, 2021-11-04:
Inductor: Thank you; I stand corrected! 😅 

> I'll add ceramic caps 2x330uF near the ESP.

👍 That should do the job!

gskjold, 2021-11-04:
As a second possibility for the issue you are describing, it might be running in to a bit of a memory issue. I have not yet exposed the available memory value in the interface, but if you open url /data.json you will find the available heap space as the value "m".

mikkel75, 2021-11-05:
I have added 2x100uF ceramic (all I could find last night)...so far the test is running smooooth...will give it a few days though..
Thanks @ArnieO for the input!

@gskjold the log says "* Free Heap RAM: 18640" which I guess should be OK...?

mikkel75, 2021-11-05:
Quitting telnet and closing webpage it increases to 20584...

gskjold, 2021-11-05:
That should be plenty. Sounds like the software has plenty to work with.

gskjold, 2021-11-07:
Attached is a firmware where you can choose UART2 in the GPIO menu.

Because Serial.swap() does not tell you which pins you are at, I had to find a different solution to support changing between them. Not 100% sure it works, but we'll see...

[Uploading firmware.zip…]()

mikkel75, 2021-11-08:
Hi Gunnar,

Thanks – but how did you implement this?
I’m pretty sure the ESP8266 only has 2 UARTs (Serial0 and Serial1). Serial1 is used for the flash chip, thus the only option is to use Serial0 on normal pins or on swapped pins (after Serial.swap()).
Isn’t it possible to simply track the number of times .swap() has been called since boot?

Thanks,
Mikkel

gskjold, 2021-11-08:
The swapped pins on UART0 is called UART2. Instead of using Serial.swap() I use Serial.pins(15,13) for UART2 and Serial.pins(1,3) for UART0. As described here: https://github.com/esp8266/Arduino/blob/986dd4209d90632917c7290537339e54ba3287d2/cores/esp8266/HardwareSerial.h#L126

mikkel75, 2021-11-08:
OK got it... Looks like the esp8266 lib uses the swap() function to choose the pins, so the result is the same :)
So that would also fail if you called the setpins with same argument twice - ie double swap and back to std pins.

Anyway - I'll give your test image a test - thanks!

mikkel75, 2021-11-08:
I'm not getting any frame / data when selecting UART2 (GPIO13) :(
Selecting GPIO13 works as before - still SW Serial I assume...
Something most not be set right for HW serial UART'2'...

gskjold, 2021-11-08:
After some reading, the swap have probably been overwritten because Serial.begin was executed after the swap. I do not currently have a ESP on hand test, but new firmware attached.

[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7495969/firmware.zip)

mikkel75, 2021-11-08:
Still no data when choosing UART2 - GPIO13 still works with soft-serial...

mikkel75, 2021-11-08:
When using UART2 the debug never shows this "Preparing for next frame" - maybe that's a hint?

gskjold, 2021-11-08:
Weird, I'll set up a full test when I have time at home.

gskjold, 2021-11-08:
Spotted another error

[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7496277/firmware.zip)

mikkel75, 2021-11-08:
YES! working now ;)

mikkel75, 2021-11-10:
Everything seems to be running smooooooth without lock-ups or missing frames ;) Please merge to master branch

BTW, have you thought about a feature to limited the MQTT TX? Like an avg over a configurable time for amps, Watt and Voltage?

gskjold, 2021-11-10:
Very good :) Changes already commited to master (https://github.com/gskjold/AmsToMqttBridge/commit/4c92e592d63a2b4d45fe47edd5b09ad6a6bf2ad6)

I have thought about reduced update rate in relations to BUS powered devices, but it does not seem to be necessary for the devices I have tested. And I have yet to encounter other situations where it would make a difference

closed 2021-11-10 by gskjold

ArnieO, 2021-11-10:
> And I have yet to encounter other situations where it would make a difference

Agreed. Does not seem necessary.

mikkel75, 2021-11-10:
One comment regarding the committed code:

I doubt this will actually work:

```
			if(pin == 3) {
				debugI("Switching UART0 to pin 1 & 3");
				Serial.pins(1,3);
			} else if(pin == 113) {
				debugI("Switching UART0 to pin 15 & 13");
				Serial.pins(15,13);
			}

```
Since, as I wrote up in this thread, the Serial.pins() function seem to call swap() and don't keep track of the number of swap()'s.
So if one sets the config to UART2 twice one would end up with UART0...or what am I forgetting...?

gskjold, 2021-11-10:
The actual implementation only execute the switch method if the rx or tx pin differs:
https://github.com/esp8266/Arduino/blob/986dd4209d90632917c7290537339e54ba3287d2/cores/esp8266/uart.cpp#L886