UtilitechAS/amsreader-firmware #141: Exception (29) ESP8266

Isaksson, 2021-11-06:
**Describe your problem**
I have flashed the latest version (1.5.6) on an Lolin D1 mini v3.1.0, I am still waiting for delivery of an M-BUS to TTL module, but I wanted just try this out to be prepared when I actual could connect this to my energy meter.

Whats happens now is that it runs for about 70 seconds then it reboots, I could see the heap memory decrease every 5 -10 seconds and the last report is around 3000 before it reboots.

Is this an issue caused because I do not have any meter connected or is this a bug in this version?

**Hardware information:**
 - Meter: -
 - AMS reader: Lolin D1 mini
 - M-bus adapter (if applicable): -

**Relevant firmware information:**
 - Version: 1.5.6
 - MQTT: no
 - HAN GPIO: -
 - Temperature sensors [e.g. 3xDS18B20] -
 - 
[Exception(29).txt](https://github.com/gskjold/AmsToMqttBridge/files/7491331/Exception.29.txt)

gskjold, 2021-11-07:
I think this could be related to not getting data from the meter like you suggested. Interesting error, I will have a try on one of mine D1 minis and see if I can figure out what is going on, thanks

Isaksson, 2021-11-07:
Thanks for that.
Little of topic but I will connect this to an Aidon Meter in Sweden that uses M-Bus communication, maybe it will work out of the box, that is going to be interesting.
I'm also very impressed of the work you have put down on this project, thanks for that.

gskjold, 2021-11-07:
It seems to me like this could be related to the meter autodetect feature. If you go to Configuration -> Meter and select your meter type and save (could be that it also throws exception at this point, but it is saved). Does the problem disappear?

ArnieO, 2021-11-07:
> Little of topic but I will connect this to an Aidon Meter in Sweden that uses M-Bus communication, maybe it will work out of the box, that is going to be interesting.

Interesting! We are actually in the process of testing a «Pow-S» prototype board on a Swedish meter right now, this morning.

https://twitter.com/amsleser/status/1451102149690241029?s=21

It would be great to get feedback on how your attempt moves forward. On Twitter or here.

gskjold, 2021-11-07:
> Thanks for that. Little of topic but I will connect this to an Aidon Meter in Sweden that uses M-Bus communication, maybe it will work out of the box, that is going to be interesting. I'm also very impressed of the work you have put down on this project, thanks for that.

Thanks for the appreciation, good to hear that people like it :) I know that there have been other swedes with M-Bus meters that have used this firmware out of the box, but like @ArnieO is mentioning there seems to be two interfaces going around in sweden.

gskjold, 2021-11-07:
Ref #40

ArnieO, 2021-11-07:
> like @ArnieO is mentioning there seems to be two interfaces going around in sweden.

I have tried to make an overview: https://github.com/ArnieO/SmartMeterDocumentation

Isaksson, 2021-11-07:
> It seems to me like this could be related to the meter autodetect feature. If you go to Configuration -> Meter and select your meter type and save (could be that it also throws exception at this point, but it is saved). Does the problem disappear?

Thanks for that, first time the uptime reaches over 70 seconds and the device does not run out of memory 
[image]

Isaksson, 2021-11-07:
Yes, its true. In Sweden there is at least three different solutions, one with RJ45 (Norwegian Standard as we call it) and then there is also the RJ12 but I think there is two versions of the protocol so even if you have the RJ12 port, you still have to verify what version of the protocol that is used.
There is a site with some information about this topic.
https://hanporten.se/ 

I have talked to the company that installed my meter, they installed it several years ago before the standard was decided but they do not have any plans to change the meters.

ArnieO, 2021-11-07:
Yes, the earliest installs (like yours) follows the Norwegian NVE standard, with M-bus. So you could use the Pow-U that I sell from my webshop www.amsleser.no.

But it is of course more interesting to construct your own device like you’re doing now!

gskjold, 2021-11-08:
Note to self: Issue possibly found. SoftwareSerial is not closed when trying different configurations for HAN port while autodetecting.

closed 2021-11-16 by gskjold