UtilitechAS/amsreader-firmware #143: Need help with testing of new firmware on multiple meter models

gskjold, 2021-11-16:
I am currently writing new code to interpret data from the meter. The new firmware should support any IEC62056-7-5 (Norwegian and Danish meters with HDLC payload) and IEC62056-21 (Swedish and possibly Dutch meters with RJ12 connector). To confirm that it actually works, I need to have it tested on multiple brands in all possible countries.

A list to start with, I will expand this as new meters are tested:

**Norway**
- [ ] Aidon 1p IT
- [ ] Aidon 3p IT
- [ ] Aidon 1p TN
- [ ] Aidon 3p TN
- [ ] Kaifa 1p IT
- [ ] Kaifa 3p IT
- [ ] Kaifa 1p TN
- [ ] Kaifa 3p TN
- [ ] Kamstrup 1p IT
- [ ] Kamstrup 3p IT
- [ ] Kamstrup 1p TN
- [ ] Kamstrup 3p TN

**Denmark**
- [ ] Kamstrup 1p IT
- [ ] Kamstrup 3p IT
- [ ] Kamstrup 1p TN
- [ ] Kamstrup 3p TN

** Sweden**
Any RJ12 meter, please let me know your setup

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/7549078/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7549080/esp8266.zip)

gskjold, 2021-11-17:
Thanks! Highly appreciated

madsbg, 2021-11-17:
Denmark, Kamstrup 3-phase with Pow-K module. How do I see if it is IT or TN model?
HAN indicator is red and debug shows: (readHanPort) Invalid HDLC, returned with -1

gskjold, 2021-11-17:
Not sure if you can see it on the meter itself, but if you have a 4 pole main breaker you have TN (400V), but if you have a 3 pole breaker it is IT or TT, which is both 230V and will result in the same datagram.

Please try the firmware below and see if that will fix your problem
[files.zip](https://github.com/gskjold/AmsToMqttBridge/files/7555290/files.zip)

madsbg, 2021-11-17:
I quess it must be TN then.

Don't know if it has anything to do with the test firmware but I went from 1.5.6 -> test firmware -> 1.5.7 and now I have lost connetion to the device. It is falshing blue but no wifi connection.

I will test the new firmware once I have recovered :slightly_smiling_face:

gskjold, 2021-11-17:
Aha, if you went back to 1.5 series firmware, it have jumped into AP mode as it does not understand the config structure from the test firmware. You need to reconfigure the device, sorry about that, should have put that up as a warning..

gskjold, 2021-11-17:
Sorry, a completely different file from same folder :)

Here you go: 
[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7555579/firmware.zip)

madsbg, 2021-11-17:
Still red HAN and same message from debug.

(readHanPort) Frame dump:
(D) 41 03 13 2F  7A E6 E7 00  DB 08 4B 41  4D 45 01 D1
(D) 22 A4 82 00  E7 30 00 0B  32 77 CE 04  24 86 23 C3
(D) 10 1C 93 EA  DB F7 69 89  26 CE E8 7D  D9 5F AE 74
(D) DC FD 92 11  01 93 FF FF  D4 2F F8 D7  32 B3 EC B9
(D) B3 05 B8 4B  0A B4 E5 8D  A9 3D 3D 28  1E E4 6B DE
(D) 9F 88 71 8F  20 3A 5F AA  D4 36 15 DD  FC C5 8A 11
(D) 76 9C E2 71  5D 56 F5 16  D5 86 EF 03  B3 AB 02 1B
(D) 93 A5 2F 50  FC 8C 3C BF  19 C4 0D 04  EE B3 E9 A2
(D) F8 0C 03 03  F5 B8 09 41  5C 59 3F B0  FC ED AF 9E
(D) 68 A8 C8 61  25 BF 42 CA  B8 8A 32 89  06 CF 14 F8
(D) 12 C1 42 CD  C0 0F E7 31  85 51 3A E0  58 48 DE 4C
(D) 57 0F A5 3A  27 5D EA 81  7D FF 6B D6  1C 24 56 74
(D) CC A5 4D 48  1D 8A A1 C1  1C 10 53 96  D3 98 F2 F3
(D) D7 DA C4 80  81 E1 81 B4  B5 81 D8 0D  42 D1 2E 85
(D) C0 DD C5 46  31 ED 07 1E  42 32 01 5D  35 5C 62 77
(D) A4 91 1C A6  4E 01 2A 4A  69 D9 38 CD  1C FC 7E
(D) (readHanPort) System title:
(D) 00 00 00 00  00 00 00 00
(D) (readHanPort) Initialization vector:
(D) 00 00 00 00  00 00 00 00  00 00 00 00
(D) (readHanPort) Additional authenticated data:
(D) 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00
(D) 00
(D) (readHanPort) Authentication tag:
(D) 00 00 00 00  00 00 00 00
(W) (readHanPort) Invalid HDLC, returned with -1

madsbg, 2021-11-17:
Maybe something to do with the meter detection? I cannot select it manually.

[image]

gskjold, 2021-11-17:
The new code does not need to know what meter type it is, only which format the data is received. I think maybe it is the format detector that is causing this. I will go throught the code and see if I can figure it out

ArnieO, 2021-11-17:
Kamstrup / TN / 3-phase in Norway:
Same issue as @madsbg

madsbg, 2021-11-17:
Looks better now. Thanks.

(I) (readHanPort) Valid HDLC, start at 47

[image]

gskjold, 2021-11-17:
Awesome! I will check why it did not detect manufacturer properly

gskjold, 2021-11-17:
Hang on, I guess you have an encrypted Kamstrup? They do not include any identifying OBIS codes, so identifying the meter manufacturer would be guesswork.

madsbg, 2021-11-17:
Yes. Data is encrypted. So maybe you could change manufacturer to: "Unknown or encrypted". At least that would make it more clear.

I am missing the {root}/meter/dlms/timestamp value. Can you bring that back at some point please?
Everything else looks fine.

gskjold, 2021-11-17:
True, I just realized when I looked through the code, have taken notes to bring that back before major release with this new code

ArnieO, 2021-11-17:
Sweet! Working now. I am on unencrypted Kamstrup, manufacturer indicated as «Unknown» here too.

madsbg, 2021-11-17:
Regarding manufacturer. I think there is a "Kamstrup_V0001" string in the data, or do you need more than that for a positive identification?

gskjold, 2021-11-17:
This is what I need, but if I remember correctly, this is not included in the encrypted datagram. It should exist in the unencrypted datagram though... Have to look closer at this

madsbg, 2021-11-17:
In the 1.5.x firmware it was published as data.lv in JSON payload. In this firmware data.lv is empty.

gskjold, 2021-11-17:
Got it, then I am mistaken. Have to look into that :)

ArnieO, 2021-11-17:
The figures do not make sense. 16A must be wrong.
I have 16 kW on L3 and 4.3 kW which is correct.

gskjold, 2021-11-17:
Can you elaborate?

ArnieO, 2021-11-17:
Sorry... my mistake. Please forget it - all is well.

madsbg, 2021-11-17:
I don't normally use {root}/meter/clock (Raw) but I just noticed, that the value is not correct (2875626203 = 2061-02-14). In JSON the clock is "t" but that value is always 0 (zero).

gskjold, 2021-11-18:
Thanks, this is something I didn't check properly.

madsbg, 2021-11-18:
Another observation: Even though everything seems to work fine with the new firmware, the Pow-K board i blinking blue-green-red-red-red.

ArnieO, 2021-11-18:
> blinking blue-green-red-red-red.

I confirm that observation. 
To be exact, it is not a triple-blink, but three single-blinks. Single-blinks signify incorrect data from the meter, so I just think this is part of the "work in progress".

Excerpt from the Pow-K user manual:
[image]

madsbg, 2021-11-18:
Thanks @ArnieO, Yes that is correct. I should have been more precise. There are about 3 sec. between each red blink.

gskjold, 2021-11-18:
Awesome, good testing everyone!
Updated first post with new firmware with the following fix:
- Package timestamp
- Meter timestamp correction
- Kamstrup meter detection

madsbg, 2021-11-18:
✔️ Red blinks are gone.
✔️ Manufacturer (Kamstrup) is now displayed correctly
✔️ "{root}/meter/clock" (Raw) seems to be correct (1637269255 = 2021-11-18 21:00:52), but I only recieved it once, so it's not conclusive. I will keep an eye on it :-)

❌ "{root}/dlms/timestamp" (Raw) and "t" (JSON) are present but are all over the place 😲
2490307205 = 2048-11-30 01:00:05
972950405 = 2000-10-31 01:00:05
4005072005 = 2096-11-30 01:00:05
1193160709 = 2007-10-23 19:31:49
2707925509 = 2055-10-23 19:31:49
3523887109 = 2081-08-31 19:31:49

madsbg, 2021-11-18:
"{root}/meter/clock" (Raw) contains the correct time when it is updated every hour on the hour, but sometimes in between an unexpected value of 0 (zero) is received.

Also "{root}/meter/import/active/accumulated" contains zero values at odd intervals.

gskjold, 2021-11-24:
Thanks for thorough testing! :) I have not had time to look at the latest reported bugs, but I will get back to it soon

Isaksson, 2021-11-24:
I noticed that my HAN communication goes from green --> yellow --> red --> green in the GUI so I connected with telnet and debug and noticed that I receives a lot of Invalid HDLC 
`(W) (readHanPort) Invalid HDLC, returned with -2` and when its valid then it looks like this
`(I) (readHanPort) Valid HDLC, start at 18`

gskjold, 2021-11-24:
If it returns with -2 it means that the checksum on the payloads header was not correct. This indicates that there is errors in the received data. 

I was also thinking it was probably that list. Since they also changed the payload, are we sure they use 8E1 parity? The document I have does not specify baud or parity for the meters with this payload.

Isaksson, 2021-11-24:
I will connect a Logic Analyzer and see what I receive.
Maybe this is the same document that you already have, but I cant seem to find the parity but they state 2400 Baudrate.

My D1 Mini crashes after 30 minute or so when running this code, It does not reboot, I have to reset it to get it up and running again. Maybe that has to do with all the checksum errors (if it works for other users with esp8266)

[aidonfd-rj45-han-interface-se-v14a-1.pdf](https://github.com/gskjold/AmsToMqttBridge/files/7596877/aidonfd-rj45-han-interface-se-v14a-1.pdf)

gskjold, 2021-11-24:
Keep an eye on "Free mem" on the main page. Maybe it is running out of memory after 30min because of some memory leak. I do not have this problem on ESP32 at least.

gskjold, 2021-11-24:
Updated first post with fix for package timestamp, some performance improvements and new graph library.

I have not yet been able to reproduce random zero meter timestamps, still searching for that one..

Isaksson, 2021-11-24:
> Keep an eye on "Free mem" on the main page. Maybe it is running out of memory after 30min because of some memory leak. I do not have this problem on ESP32 at least.

I was looking at the GUI when the device freezes and there was Free mem: 18.2kb, I switched to your latest snapshot and now I could select my main fuse size :) thanks for that.

gskjold, 2021-11-24:
18.2 kib sounds reasonable, so it must be something else. I will try to reproduce it with a d1 mini

Isaksson, 2021-11-24:
Have you changed something with the buffer in the latest snapshot?
Sometimes I see this when debugging
`(I) (readHanPort) Buffer overflow, resetting`

gskjold, 2021-11-24:
Buffer is same size (1024 bytes), but it fills the buffer in a different way. If there is errors in the received data, it may continue to fill the buffer until it is full and reset.

madsbg, 2021-11-24:
> Updated first post with fix for package timestamp, (...)

I have updated to the latest version (I think) but now "{root}/dlms/timestamp" (Raw) is missing and "t" (JSON) is always 0 (zero).

BTW: Could you perhaps give these beta firmware versions a visible build number in the GUI? That would make it easier to refer to a specific version.

Isaksson, 2021-11-25:
> 18.2 kib sounds reasonable, so it must be something else. I will try to reproduce it with a d1 mini

With the latest build the device have not yet frozen, now it reboots instead.

Isaksson, 2021-11-26:
Would it be possible to add a debug option so that it prints out the entire message retrieved from the HAN port to telnet debug?
It would be nice to see whats happening on most of my communication if its an actual error from my Mbus to ttl or if the error is that the code cant parse the information.

gskjold, 2021-11-26:
If you go to System -> Debugging and set the debug level to "Debug" it will dump the entire frame received in telnet as well as serial.

Isaksson, 2021-11-26:
> If you go to System -> Debugging and set the debug level to "Debug" it will dump the entire frame received in telnet as well as serial.

Yes, of course  :) I have seen this many times know, it just flow above my head.

I notice that there is some different events.

(W) (readHanPort) Invalid HDLC, returned with -2
(I) (readHanPort) Buffer overflow, resetting
(I) (readHanPort) Valid HDLC, start at 18
and also nothing reported at all in the debug after a reading and as you suggested its added to the buffer (but no info in debug) and after a couple of readings it gets overflow, but the question is, do you think that this is because of some "trash" from the TTL out or is it because the code cant decode the data?
That my Swedish Meter is sending some different data?

gskjold, 2021-11-27:
Got a treat for you tonight; first post updated with new firmware with the following updates:
- Improved new graphs
- Added usage graph for last 24hrs and last month
- Added loading indicator to all uploads
- Added more information on firmware upload page
- Added possibility to specify voltage divider for Vcc in GPIO menu
- Added ADC reading for Vcc when possible (ESP32)

ArnieO, 2021-11-27:
> Got a treat for you tonight

Pre-Christmas candy! 🎅

madsbg, 2021-11-27:
I like the improvements to the firmware upload experience 👍
I also like the usage graphs. Maybe you could also add a measurement unit?
[image]

Could you please send the package timestamp with every MQTT update like the 1.5.x versions?
Thanks 😃

gskjold, 2021-11-28:
A type-o have caused package timestamp to be missing, will fix it :)

gskjold, 2021-11-28:
A couple of questions:

1. Looking at the graph for daily usage, what is most intuitive; 00-23 or 01-24 ? Right now you will see that it shows use for previous hour. Right now at 1300 CET the graph ends at 12 which has the value for used kWh between 1200-1300.

2. In meter configuration I have removed the dropdown for choosing meter model. The main reason for this is that I hope the coming versions will support meters from many countries and the number of meters and associated configuration in some countries are staggering. But the baud/parity/invert configuration is also very alien for most people, so what would be the best way to solve this?

madsbg, 2021-11-29:
1. To me 00-23 is the most logical. Now the time is 10:05 and the graph ends at 9. I would expect it to end at 10 but with the graphs only drawn to 9. Does that make sense?
[image]

2. I'm not sure what you mean? Do you want the hide the settings or maybe separate them in another section?
[image]

ArnieO, 2021-11-29:
> EDIT: Changed my mind regarding 1. 😃

Hehe, this is exactly why the question is raised! 
At first glance it looks counter intuitive that the data point created at 10:00 is labeled 9. 
But that data point is for the time period where the clock is 9:xx, so I think it could be seen as the most "correct" way to do it.

madsbg, 2021-11-29:
I think a data point should hold the data form the previous time period.

ArnieO, 2021-11-29:
> The same with the monthly graph. It should end at 29 and not 28 because today is 29. Or am I misunderstanding?

I disagree, as the data point "28" is the consumption yesterday (28th Nov). One data point is created each midnight. 
What you mean is maybe that there should be a new point 29 already, that increases during the day today?

madsbg, 2021-11-29:
If there is only one data point that holds the accumulated value for the entire time period, then I think bar charts would be more appropriate.

gskjold, 2021-11-29:
I hear what you are saying about the bar chart. Here is an example of that with relevant data:
[image]

ArnieO, 2021-11-29:
Yeah... I think this is better! 👍 
And I would like to see the measurement values as you did for voltage and current ("kWh" not needed, it is already on the axis).
And why not repeat the purple color you use in the menus?

madsbg, 2021-11-29:
> I hear what you are saying about the bar chart. Here is an example of that with relevant data:

I like it. That looks really good.

gskjold, 2021-11-29:
Ooh I like the color idea!
[image]

[80 more replies]

closed 2021-12-10 by gskjold