UtilitechAS/amsreader-firmware #144: ESP8266 hung on boot

aadnehovda, 2021-11-17:
**Describe the bug**
After modifying some MQTT parameters and hitting Save, the device stopped responding and fails to boot. Does not respond to the AP-mode button so I cannot reset through GUI. Erasing and reflashing (tried both 1.5.6 and 1.5.7) solved the issue. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Serial output during boot:

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
   Sensors: 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
   Sensors: 0
```

**Hardware information:**
 - Meter: Kamstrup 3P-IT
 - AMS reader: Pow-K

**Relevant firmware information:**
 - Version: 1.5.6
 - MQTT: yes

**Additional context**
I've attached a full flash dump. I reflashed using this dump and it still failed to boot.
[ams2mqtt-esp8266-1.5.6-failed_boot.bin.zip](https://github.com/gskjold/AmsToMqttBridge/files/7556933/ams2mqtt-esp8266-1.5.6-failed_boot.bin.zip)

gskjold, 2021-11-17:
Thanks! I will have a look at it when I have a available time slot

gskjold, 2021-12-12:
Alright, worry this have taken some time, but I did find the problem. Although the value was logged to serial, dividing zero with a number was something that could cause an error apparently. I am not sure why... 
https://github.com/gskjold/AmsToMqttBridge/blob/ff935fc92096e80a5ba72f5ccb8d2c032c986137/src/AmsConfiguration.cpp#L1015

closed 2021-12-12 by gskjold