UtilitechAS/amsreader-firmware #145: Ampere is shown as max current of the phases

henriklund, 2021-11-18:
I have noticed that Ampere shown in the GUI is simply a max. value of each of the available (I1 - I3) currents. This seems counterintuitive since shown effect (left hand gauge - read from meter) divided by average voltage (center gauge) does match this value.

I would have expected to see a result matching I = W / V.

ArnieO, 2021-11-18:
The idea is that it is the highest current that might make the fuse trip, so that is displayed grapgically. As this is three phase, there is not a simple direct relation between voltage, current and power, due to the phase difference.

But you are not the first to find this display counter intuitive, so: In your opinion; what is the best way to present these three data sets?

henriklund, 2021-11-19:
The fuse constraint was also one of the caveats that I thought about  when posting this, and if there is no simple relation between the three values, I agree that a simple W/V wouldn't make sense.
What could nice then is some sort of visualization that took this complexity into account. One idea could be to present an alternative display of (at least) the amp gauge. Maybe something that could look like: 
[image]
(I've fiddled a bit around with the gaugeMeter.js to achieve this for 1P, 2P and 3P)

ArnieO, 2021-11-20:
I think this is a great idea! It definitively gets my vote.

henriklund, 2021-11-20:
I have attached the gaugemeter.js I have been playing around with. I need someone to review the code + tighten it up (I am def. not a coder). I tweaked it so that the color each of the three measures will change individually (previously all three gauges would have same color). I am not certain which of the two alternatives is better.
[image]

[gaugemeter-new.zip](https://github.com/gskjold/AmsToMqttBridge/files/7574481/gaugemeter-new.zip)

gskjold, 2021-11-20:
Thanks, I will have a look :)

ArnieO, 2021-11-20:
> I am not certain which of the two alternatives is better.

I think this looks great - but this boils down to individual preference.

gskjold, 2021-11-24:
The solution is not bad and would have worked out. However, I am currently adding some new nice features which requires another type of graph, so I had to import a new graph framework for that. Not wanting to keep two frameworks, I have dropped gaugemeter.js... Here is a new solution for volt and amp:

[image]

madsbg, 2021-11-24:
I like the new graphs 👍

henriklund, 2021-11-24:
That should do the trick :)

ArnieO, 2021-11-24:
Easy to read and understand! 👍

Isaksson, 2021-11-24:
This looks really good, much better than the current gauges.
But, when presenting the Amps with percentage then its important to add some more options for the "Main fuse"
16A and 20A needs to be added to cover most of the users in Sweden.

gskjold, 2021-11-24:
Alright, will do :)

closed 2021-11-24 by gskjold