UtilitechAS/amsreader-firmware #146: Incorrect frame format 10 00 16

Isaksson, 2021-11-24:
**Describe the bug**
Incorrect frame format 10 00 16

**Hardware information:**
 - Meter: Aidon (Sweden)
 - AMS reader: ESP8266 LolinD1 Mini
 - M-bus adapter TSS721A Module M-bus to TTL

**Relevant firmware information:**
 - Version: 1.5.7
 - MQTT: yes
 - HAN GPIO: GPIO5
 - Temperature sensors: -

**Additional context**
I have finally received my MBus to TTL converter and I connected it to my Aidon Meter (Sweden), turned on debugging in AMS Reading and this is the result that I get from the HAN port.

`(D) Incorrect frame format 10
(D) 00 16
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (AmsWebServer)Serving /data.json over http...
(D) (loop) Used 0 ms to update temperature
(D) (AmsWebServer)Serving /data.json over http...
(D) Buffer overflow
(D) 7E A2 43 41  08 83 13 85  EB E6 E7 00  0F 40 00 00
(D) 00 00 01 1B  02 02 09 06  00 00 01 00  00 FF 09 0C
(D) 07 E5 0B 18  03 0C 04 28  FF 80 00 FF  02 03 09 06
(D) 01 00 01 07  00 FF 06 00  00 07 74 02  02 0F 00 16
(D) 1B 02 03 09  06 01 00 02  07 00 FF 06  00 00 00 00
(D) 02 02 00 16  1B 02 03 09  06 01 00 03  07 00 FF 06
(D) 00 00 00 00  02 02 0F 00  16 1D 02 03  09 06 01 00
(D) 04 07 00 FF  06 00 00 02  DA 02 02 0F  00 16 1D 02
(D) 03 09 06 01  00 1F 07 00  FF 10 00 09  02 02 0F FF
(D) 16 21 02 03  09 06 01 00  33 07 00 FF  10 00 13 02
(D) 02 0F FF 16  21 02 03 09  06 01 00 47  07 00 FF 10
(D) 00 3D 02 02  0F FF 16 21  02 03 09 06  01 00 20 07
(D) 00 FF 12 09  0C 02 02 0F  FF 16 23 02  03 09 06 01
(D) 00 34 07 00  FF 12 09 1C  02 02 0F FF  16 23 02 03
(D) 09 06 01 00  48 07 00 FF  12 09 0B 02  02 0F FF 16
(D) 23 02 03 09  06 01 00 35  07 00 FF 06  00 00 00 DB
(D) 02 02 0F 00  16 1B 02 03  09 06 01 00  16 07 00 FF
(D) 06 00 00 00  00 02 02 0F  00 16 1B 02  03 09 06 01
(D) 00 17 07 00  FF 06 00 00  00 00 02 02  0F 00 16 1D
(D) 02 03 09 06  01 00 18 07  00 FF 06 00  00 00 39 02
(D) 02 0F 00 16  1D 02 03 09  06 01 00 29  07 00 FF 06
(D) 00 00 01 81  02 02 0F 00  16 1B 02 03  09 06 01 00
(D) 2A 07 00 FF  06 00 00 00  00 02 02 0F  00 16 1B 02
(D) 03 09 06 01  00 2B 07 00  FF 06 00 00  00 00 02 02
(D) 0F 00 16 1D  02 03 09 06  01 00 2C 07  00 FF 06 00
(D) 00 01 05 02  02 0F 00 16  1D 02 03 09  06 01 00 3D
(D) 07 00 FF 06  00 00 05 4B  02 02 0F 00  16 1B 02 03
(D) 09 06 01 00  3E 07 00 FF  06 00 00 00  00 02 02 0F
(D) 00 16 1B 02  03 09 06 01  00 3F 07 00  FF 06 00 00
(D) 00 00 02 02  0F 00 16 1D  02 03 09 06  01 00 40 07
(D) 00 FF 06 00  00 01 88 02  02 0F 00 16  1D 02 03 09
(D) 06 01 00 01  08 00 FF 06  03 B1 34 49  02 02 0F 00
`

Is this information enough for you to make a guess what's wrong :) or do I need to provide more information?

ArnieO, 2021-11-24:
Swedish AMS meters are currently only supported if they follow the "Norwegian standard" (NVE). 

See my overview here for more information: https://github.com/ArnieO/SmartMeterDocumentation

If the connector is RJ45 (an ethernet cable fits) you should be good.

Isaksson, 2021-11-24:
> Swedish AMS meters are currently only supported if they follow the "Norwegian standard" (NVE).
> 
> See my overview here for more information: https://github.com/ArnieO/SmartMeterDocumentation
> 
> If the connector is RJ45 (an ethernet cable fits) you should be good.

This is a meter that was installed before the Swedish Standard was in place so this is RJ45 and Mbus solution, we had a discussion about this when i noticed a bug in the code.
https://github.com/gskjold/AmsToMqttBridge/issues/141

ArnieO, 2021-11-24:
Ah... ok  - sorry for not having recognized you!

How is your board powered? Have you tried adding capacitance?

gskjold, 2021-11-24:
Could I tempt you to test the latest firmware in #143 ?

Isaksson, 2021-11-24:
> Ah... ok - sorry for not having recognized you!
> 
> How is your board powered? Have you tried adding capacitance?

The Mbus board (on the Mbus side) is powered from the Master and the TTL part of the board is powered from the 3.3v from the D1 Mini (this could be a problem), I have just hooked this up so I have not tested much at all, wanted to post here first and start a discussion :)

Isaksson, 2021-11-24:
> Could I tempt you to test the latest firmware in #143 ?

Sure, I will do that right away and post the result.

gskjold, 2021-11-24:
It is most likely because of this: 
(D) Buffer overflow
and the buffer is larger on the test version.

Also, this seems to be a longer list than the Norwegian payload (27 vs 18 items)

Isaksson, 2021-11-24:
Yes, you are correct.
It works with the new code.
[image]

Thanks for that

gskjold, 2021-11-24:
Awesome! :) If you discover any bugs in that, comment in #143

closed 2021-11-24 by gskjold