UtilitechAS/amsreader-firmware #147: Possibility to trigger factory reset on boot

gskjold, 2021-11-27:
**Describe the solution you'd like**
Add a grace period on boot, waiting for AP button hold. Wait 2 seconds, if AP button is pressed, full factory reset.

**Additional context**
If a user accidentally bricks a device because of unknown limitation in config input, full reset should be possible.

ArnieO, 2021-11-27:
Can you please describe the intention a bit more extensively? First glance looks like a potential for accidental full factory reset.

gskjold, 2021-11-27:
As we have seen, the firmware can reach critical state by malconfiguration. There should be some way of recovering without FTDI.

Could also go straight to AP mode and hopefully that works.
I am open for suggestions here.

ArnieO, 2021-11-27:
OK, intention understood. But will the µC reboot as normal before crashing if bricked in that way?

My concern is that if this is active at any reset/reboot, there is a risk for unintended accidental factory reset.

If implemented, I suggest starting with a 10(?) second period with some kind of LED signal alerting the user ("Do you really want to do a factory reset?", e.g. switching between red and yellow 500 ms each), were a new press of RESET during that period will stop the process and give a normal reboot. Or maybe better: Reconfirm during that period with new press on AP button.

gskjold, 2021-11-27:
Cool idea, I will try that :)

madsbg, 2021-11-27:
How about only resetting when the AP button is pressed and held for at least 10 sec. during power-on? That would prevent any unintentional resets.

sveip, 2021-11-27:
Better to check for long button press, if pressed for 10s during boot, rename configuration and load backup configuration. Maybe better to have a successful start flag after config is loaded, and if this flag isn't set (meaning last boot failed/hanged) load backup config during next boot.

bardahlm, 2021-12-20:
Can a feature like this be coupled with some kind of boot helper that automatically reconfigures the device if it resets? Would probably require that network config is kept and not cleared.

gskjold, 2022-02-19:
I think I have landed a solution. Only if AP button is configured:
1. Green LED will light for 1s. Start holding AP button.
2. If AP button is active after 1s, show yellow LED. 
3. Keep holding button
4. If AP button is active after 2s, show red LED.
5. Release button if you want to reset, keep holding if you regret.
6. If AP button is released, reset to factory defaults and continue boot and confirm with 6 red blinks

This way, if there is a configuration error where AP button seems to be active all the time, it will not reset to factory.

For dev boards, LED will be as following:
1. On
2. Flash 4 times, then off
3. Keep holding
4. On
5. Release.
6. Flash 6 times to confirm factory reset

closed 2022-02-20 by gskjold