UtilitechAS/amsreader-firmware #152: Error saving configuration

manfi001, 2021-12-04:
Flashed firmware to a ESP07 with ESP-Flasher
device starts in AP mode after reboot but when trying to save initial configuration, I get error msg: Error saving configuration!
what could be wrong?
 - M-bus adapter: TSS721

**Relevant firmware information:**
 - Version: [e.g. 1.5.8]
 - MQTT: [yes/no]
 - HAN GPIO: [not yet connected ]
 - Temperature sensors [none]

gskjold, 2021-12-04:
Sorry, but the project has been built with a 4MB partition layout which overflows the 1MB memory chip on ESP07

manfi001, 2021-12-04:
Thanks for info.
Flashed to D1mini and o.k.
Trying to use it in Austria with a Sagemcom T210-D smart-meter.
Will get the decryption keys shortly and let you know how it works.
br

closed 2021-12-04 by manfi001

gskjold, 2021-12-04:
Awesome! Let me know how that works out, I would really like to get this working for as many meters as possible.

gskjold, 2021-12-04:
By the way, your meter will only be supported by the 2.0 series firmware we are currently testing in #143

manfi001, 2021-12-04:
Hi Gunnar,If you are interested, I could send you the documentation about the meter interface of my electricity provider. however it is in GermanbrSent from my Tablet

gskjold, 2021-12-04:
Thanks but I have found some information on it already. It uses ESMR 5.0, which technically should work with v2.0.