UtilitechAS/amsreader-firmware #153: ESP8266 WiFi reconnect not working with MQTT

gskjold, 2021-12-04:
**Describe the bug**
If MQTT is enabled and the device is disconnected from WiFi, it is unable to reconnect.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up device with MQTT
2. After MQTT is connected, go to your WiFi controller or AP and disconnect the device
3. Observe that it will not reconnect to WiFi

**Hardware information:**
D1 mini, and probably all ESP8266

**Relevant firmware information:**
 - Version: 2.0.0
 - MQTT: yes

madsbg, 2021-12-04:
I can confirm that Pow-K (ESP8266) with software v2.0-rc4 and MQTT enabled fails to reconnect after a manual/forced disconnect (deauth) from the access point. After the automatic reboot, it reconnects successfully.

These are some of the error messages from the access point: (BTW, I am not a wireless communication expert)

Dec  4 21:43:08.465 CET: (0000.0000.0000): dot11_dot1x: [handshake retry count = 3] Sent WPAv2 PTK msg 3 to client, no timer set
Dec  4 21:43:08.465 CET: (0000.0000.0000): dot11_dot1x: [handshake retry count = 3] Handshake msg to client, timer set: timeout 1000 ms
Dec  4 21:43:09.465 CET: (0000.0000.0000): DOT1X_SM: Executing Action [state: WPAV2_PTK_MSG4_WAIT, event: TIMEOUT] for client
Dec  4 21:43:09.465 CET: (0000.0000.0000): dot11_dot1x: WAPv2 M4 not received after max retries, deauthing the client
Dec  4 21:43:09.465 CET: (0000.0000.0000): dot11_dot1x: 4-way Handshake failure for client
Dec  4 21:43:09.465 CET: %DOT11-7-AUTH_FAILED: Station e8db.84xx.xxxx Authentication failed

With MQTT disabled, the module reconnects to wifi within a couple of seconds after a forced disconnect.

gskjold, 2021-12-04:
I dont understand what I did... I've tried reverting to arduino, espidf and lib versions that was used in 1.5.8 (which does not have the problem), but its still a problem for 2.0. It must be something I have changed, but I cant understand what it is yet

madsbg, 2021-12-04:
I wish I could help, but I'm not a programmer. It is odd that the issue seems to be connected to MQTT.

I noticed that enabling/disabling MQTT requires a restart to take effect. Was that the same with v1.5.x?

gskjold, 2021-12-04:
I tried 1.5.7 and 1.5.8, and both recover from disconnect from wifi with mqtt configured. Which made me think it was related to esp/arduino version, but it seems like it it something I changed. I just have to figure out what it is 🙂

gskjold, 2021-12-05:
Added issue https://github.com/256dpi/arduino-mqtt/issues/274

gskjold, 2021-12-06:
Fixed with commit https://github.com/gskjold/AmsToMqttBridge/commit/bc6d45ecf2bff127db973dc98f78c7ce0f81bfc1
I have no idea why, but now MQTT SSL is not supported on ESP8266

closed 2021-12-06 by gskjold