UtilitechAS/amsreader-firmware #156: Swedish Aidon meters show 10x of actual value

gskjold, 2021-12-10:
Implement interpretation of the scale and unit fields

Isaksson, 2021-12-10:
Here is one frame from debugging.

`(D) (readHanPort) Frame dump (581b):
(D) 7E A2 43 41  08 83 13 85  EB E6 E7 00  0F 40 00 00
(D) 00 00 01 1B  02 02 09 06  00 00 01 00  00 FF 09 0C
(D) 07 E5 0C 0A  05 10 39 00  FF 80 00 FF  02 03 09 06
(D) 01 00 01 07  00 FF 06 00  00 07 E5 02  02 0F 00 16
(D) 1B 02 03 09  06 01 00 02  07 00 FF 06  00 00 00 00
(D) 02 02 0F 00  16 1B 02 03  09 06 01 00  03 07 00 FF
(D) 06 00 00 00  00 02 02 0F  00 16 1D 02  03 09 06 01
(D) 00 04 07 00  FF 06 00 00  02 48 02 02  0F 00 16 1D
(D) 02 03 09 06  01 00 1F 07  00 FF 10 00  09 02 02 0F
(D) FF 16 21 02  03 09 06 01  00 33 07 00  FF 10 00 25
(D) 02 02 0F FF  16 21 02 03  09 06 01 00  47 07 00 FF
(D) 10 00 2E 02  02 0F FF 16  21 02 03 09  06 01 00 20
(D) 07 00 FF 12  08 E3 02 02  0F FF 16 23  02 03 09 06
(D) 01 00 34 07  00 FF 12 08  D8 02 02 0F  FF 16 23 02
(D) 03 09 06 01  00 48 07 00  FF 12 08 DF  02 02 0F FF
(D) 16 23 02 03  09 06 01 00  15 07 00 FF  06 00 00 00
(D) D5 02 02 0F  00 16 1B 02  03 09 06 01  00 16 07 00
(D) FF 06 00 00  00 00 02 02  0F 00 16 1B  02 03 09 06
(D) 01 00 17 07  00 FF 06 00  00 00 00 02  02 0F 00 16
(D) 1D 02 03 09  06 01 00 18  07 00 FF 06  00 00 00 36
(D) 02 02 0F 00  16 1D 02 03  09 06 01 00  29 07 00 FF
(D) 06 00 00 03  0C 02 02 0F  00 16 1B 02  03 09 06 01
(D) 00 2A 07 00  FF 06 00 00  00 00 02 02  0F 00 16 1B
(D) 02 03 09 06  01 00 2B 07  00 FF 06 00  00 01 21 02
(D) 02 0F 00 16  1D 02 03 09  06 01 00 2C  07 00 FF 06
(D) 00 00 00 00  02 02 0F 00  16 1D 02 03  09 06 01 00
(D) 3D 07 00 FF  06 00 00 03  F9 02 02 0F  00 16 1B 02
(D) 03 09 06 01  00 3E 07 00  FF 06 00 00  00 00 02 02
(D) 0F 00 16 1B  02 03 09 06  01 00 3F 07  00 FF 06 00
(D) 00 00 00 02  02 0F 00 16  1D 02 03 09  06 01 00 40
(D) 07 00 FF 06  00 00 00 E9  02 02 0F 00  16 1D 02 03
(D) 09 06 01 00  01 08 00 FF  06 03 C2 5A  64 02 02 0F
(D) 00 16 1E 02  03 09 06 01  00 02 08 00  FF 06 00 00
(D) 00 00 02 02  0F 00 16 1E  02 03 09 06  01 00 03 08
(D) 00 FF 06 00  04 5D 06 02  02 0F 00 16  20 02 03 09
(D) 06 01 00 04  08 00 FF 06  00 B4 9D 89  02 02 0F 00
(D) 16 20 1C 90  7E
(I) (readHanPort) Valid HDLC, start at 18
`

gskjold, 2021-12-10:
Try this one:
[firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7694391/firmware.zip)

Isaksson, 2021-12-10:
> Try this one: [firmware.zip](https://github.com/gskjold/AmsToMqttBridge/files/7694391/firmware.zip)

Now it looks much better :) its the correct total kWh

gskjold, 2021-12-10:
Great! Will be included in next patch release

closed 2021-12-10 by gskjold