UtilitechAS/amsreader-firmware #157: v2.0.0 fails HAN i/f with Invalid HDLC

mikkel75, 2021-12-10:
**Describe the bug**
Updated from 1.5.8 which was working fine to 2.0.0, now I get continious HAN error:

(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 AE 90 F6
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 AE 90 F6  00 10 C1 48
(D) (readHanPort) Additional authenticated data:
(D) 30 3B FF 3F  8A 12 10 40  8A 08 10 40  01 00 00 00
(D) 18
(D) (readHanPort) Authentication tag:
(D) F9 F5 48 8C  3D 64 9E 31  DA BC F0 45
(W) (readHanPort) Invalid HDLC, returned with -91
(D) (AmsWebServer)Serving /data.json over http...

**Hardware information:**
 - Meter: Kamstrup DK
 - AMS reader: ESP8266 GPIO13

**Relevant firmware information:**
 - Version: 2.0.0
 - MQTT: yes
 - HAN GPIO: GPIO13 (UART2)

gskjold, 2021-12-10:
Could you give me the first two lines of the full package? And also the number for bytes

mikkel75, 2021-12-10:
D) (AmsWebServer)Serving /data.json over http...
(D) (readHanPort) Frame dump (491b):
(D) 7E A1 E9 41  03 13 C6 37  E6 E7 00 DB  08 4B 41 4D
(D) 45 01 AE 90  F6 82 01 D0  30 00 10 C1  AC C1 83 B3
(D) 8E E5 72 2C  A7 8E 95 73  F7 56 4A 15  0B 02 44 BD
(D) 49 73 27 9E  8D F1 AF BD  84 3B 53 10  ED 14 55 30
(D) 06 10 1C 16  D5 CE 24 4B  E0 1B B8 4C  8C 0D 07 B7
(D) 2D 64 78 7F  22 82 2F F6  2C CF B4 C8  5F 03 AA 2F
(D) 55 D7 C3 D9  DA 97 F9 E3  4E C3 FF 4D  9B D5 0C 9D
(D) 08 AA 24 37  0C 41 EA 7E  B9 B9 C5 88  34 94 80 87
(D) C6 AA EE BB  36 AB 43 89  C1 9C C4 F4  BB 08 CF C9
(D) 57 D7 F9 F3  97 52 AE E2  1B E4 CA 0A  3C D8 64 47
(D) CB EB 92 E0  5B B0 32 ED  55 BB F5 EE  4F 59 65 0A
(D) 87 F7 95 D1  25 7D A7 2D  18 1E 87 C1  BB CE C5 4C
(D) E7 C3 0C 4C  DB 66 AB F5  C3 67 91 B2  16 70 E3 EF
(D) C6 64 58 AF  B1 42 01 8C  E9 4D 3F DF  9D 24 E5 50
(D) A9 E0 07 49  D5 F9 DD 1A  CF C3 CE D3  61 B9 DF D9
(D) 9D 38 D1 BD  53 56 25 C9  81 50 DB D0  ED 79 86 98
(D) FC 60 F7 96  46 97 2E 1A  2F 64 43 6F  65 C8 EC FD
(D) 0C 50 3B 2F  77 97 9F A9  96 C5 A8 87  EB EA C2 7D
(D) B4 51 AF 9E  B7 59 31 BD  C4 BD 0F 7C  E9 19 97 D5
(D) E3 90 23 73  65 C5 C4 87  E3 7A E2 EB  55 05 61 74
(D) 63 4A D2 FE  6F D1 F3 97  8F 99 F9 4A  3E 6E D1 2C
(D) 0E 1E AA 57  3D 7D 9A 4B  F6 C7 45 20  7F 88 90 60
(D) B5 54 D2 9C  A8 D4 11 C3  BD 6B 04 74  76 71 C5 6A
(D) 1A A0 7E 7E  E1 2A 1A 54  FB 5E 52 50  20 10 79 97
(D) D4 19 0A E7  E6 EC AA 6F  4E BB DE 4B  7A 88 08 3F
(D) CE 3D 00 94  0D 97 23 33  D7 9F 69 26  47 39 1F 62
(D) 26 CF 6C E4  A4 46 F1 6E  1F 3D 62 91  DD C4 47 35
(D) B9 A4 2D F3  76 F2 D1 B0  59 CF 79 BC  B8 01 BD D7
(D) EB 85 3D 2B  D4 6A 8E 0A  3E 5A CD 82  42 E3 F5 06
(D) 4C 4C 90 D1  68 60 97 4C  ED 17 F9 93  59 45 42 60
(D) 7C 76 C0 71  29 87 5B 03  72 5F 7E
(D) (readHanPort) System title:
(D) 4B 41 4D 45  01 AE 90 F6
(D) (readHanPort) Initialization vector:
(D) 4B 41 4D 45  01 AE 90 F6  00 10 C1 AC
(D) (readHanPort) Additional authenticated data:
(D) 30 3B FF 3F  8A 12 10 40  8A 08 10 40  01 00 00 00
(D) 18
(D) (readHanPort) Authentication tag:
(D) 59 45 42 60  7C 76 C0 71  29 87 5B 03
(W) (readHanPort) Invalid HDLC, returned with -91

gskjold, 2021-12-10:
Btw, can you check in config and confirm that your encryption key and authentication key is correct?

mikkel75, 2021-12-10:
Ahhh sorry was the keys that didn't 'survive' the update...values was still present - just not the right ones....

closed 2021-12-10 by mikkel75

gskjold, 2021-12-10:
Alright, good to know. This means that there is probably an issue in the update process, I will have a look at that, thanks!

mikkel75, 2021-12-10:
New GUI looks fantastic! Great job!