UtilitechAS/amsreader-firmware #158: Serial data stops working when 1-wire tempsensor is activated

kng, 2021-12-11:
**Describe the bug**
Enabling temperature sensor messing up HAN port.

**To Reproduce**
1. factory reset and initial setup for pins and mqtt.
2. everything works fine.
3. connecting a ds18b20 and configuring it's pin, temperature shows up ok.
4. HAN port stops receiving data. Enabling the telnet debug shows a few corrupted packets.
5. Remove the temp sensor, changes nothing.
6. completely disabling 1-wire not possible, only changing the pin.
7. only a factory reset and not setting 1-wire pin solves the issue.

**Expected behavior**
Enabling 1-wire should not conflict with HAN port.
Disabling the 1-wire should be possible, not just changing it's pin.

**Hardware information:**
 - Meter: Kamstrup
 - AMS reader: custom esp8266 with lipo-battery

**Relevant firmware information:**
 - Version: 2.0.0
 - MQTT: yes
 - HAN GPIO: GPIO13
 - Temperature sensors 1xDS18B20 on GPIO14

gskjold, 2021-12-12:
I will try to reproduce this locally. I'm assuming you have HAN set to UART2?

gskjold, 2021-12-12:
Tried a few different setups, but I am not able to reproduce this problem..

Just a heads up on the debugging, if you run DEBUG level with telnet, the logging may use too much time and could cause the serial buffer to overflow and cause invalid hdlc.

kng, 2021-12-12:
HAN port is set to UART2, yes. 
Also only enabled the telnet debugging after the problem was showing up.
Could it be something else that is unset in the gpio settings that gets a value saved ? 
I have observed some of them gets different default values, like AP button = 0 sometimes.
I would be helpful to be able to disable these if they're not in the hw. Like pin = [disabled|1|2|...]

I did notice in the older fw (snapshot or 1.5) that the temperature polling collided with the serial receive. I could see the messages in mqtt getting closer in sync (han msg every 10 sec, temperature 11 or 9) and when they overlapped it corrupted the han message. It should probably be triggered by a recent han packet, or otherwise free running.

[image]

mikkle, 2021-12-16:
For what it's worth, I had no success reading anything on the HAN port on my Pow-K (Kamstrup Omnipower, Encrypted, DK) with releases 2.0.0 and 2.0.1. Checksum error on all received frames.
Commit 5408b3c2a9b1fd7d7a0c52a806473f9dc8daddbf "Make sure temperatures are read just after receiving HAN data" fixed that for my case at least.

gskjold, 2021-12-18:
@kng Did this commit fix it for you too?

kng, 2021-12-18:
I'm running on snapshot now and had to change the 1-wire pin to GPIO4, for some reason GPIO14 messes things up on my board, even if unconfigured.
The temperature reads ok and just after every two meter messages.

gskjold, 2021-12-18:
Alright cool. Before it updated every 15s. Since we changed the code and you have  Kamstrup meter that sends data every 10s, it updates every other message. I don't think that makes much difference for anyone, the temp normally doesnt change that much in 20s anyway.

kng, 2021-12-18:
I just had to disable the separate temperature message, it caused a lot of warnings in the log for missing elements etc. I could not fix it with the same solution as I used for the tPI message. I should probably work on the HA payload option with separate topics.

closed 2021-12-18 by kng