UtilitechAS/amsreader-firmware #159: Home Assistant: MQTT Discovery

kng, 2021-12-11:
Option to publish MQTT messages that describes available sensor data, for easy integration in hass.io
Adding this as a payload type of Home Assistant, or similar.

https://www.home-assistant.io/docs/mqtt/discovery/

Also making the important ones to have state_class of measurement, total_increasing or total, so the Energy Dashboard gets the proper data.

gskjold, 2021-12-13:
I do not have HA myself, so it would be better if someone who has HA at least provided a full list of changes needed to the data that we already supply to MQTT

kng, 2021-12-13:
I'm researching and experimenting with it currently, will report back.  I'm no good at c++ thou, perhaps giving it a try anyways.

gskjold, 2021-12-14:
The implementation is not so important, but rather the format for the payload that needs to be sent to MQTT.

aadnehovda, 2021-12-14:
I had a look at this a little while back, I'll try to recollect some of the config I tested. Afaict, there's support for one HA sensor per MQTT topic, so using a JSON payload will probably not make much sense, since that's one topic and we'd want more sensors. The various topics have to follow a specific pattern. The deep hierachy of topics in the current full MQTT output must be flattened out. 

It would be nice to have the accumulated import/export values sent after each ams packet and not just every hour. That should make realtime data aggregation in HA easier since you can use the total_increasing sensor type directly instead of the kW power samples and have HA integrate those values.

It's also possible to push metadata like sw_version, rssi, voltage, etc for the meter itself and the ams reader (as a via_device). Not sure how important it is to keep the amount of traffic low, these config blocks rarely change so perhaps they can be cached and only resent if there is change, probably multiple ways to optimize that I'm sure.

kng, 2021-12-15:
I'm trying to use the json format without any changes as for now, and it is working.
For HA I think it should be split into two topics, one for the device status and one for the HAN data.
Basically move the "data" : { .. } to a separate topic.
There is also a quirk with missing values (tPI, tPO etc), but this can be handled. 
I opted for the unique ID that Tasmota uses, like the last three bytes of the mac adress, in this example: 78ABCD

This is what I used to get it started:
mqtt topic: "homeassistant/sensor/78ABCD_status/config"
`{
  "name" : "AMS reader status",
  "stat_t" : "tele/ams_78ABCD/sensor",
  "json_attr_t" : "tele/ams_78ABCD/sensor",
  "unit_of_meas" : "dB",
  "val_tpl" : "{{value_json['rssi']}}",
  "ic" : "mdi:information-outline",
  "uniq_id" : "78ABCD_status",
  "dev" : {
    "ids" : [ "78ABCD" ],
    "name" : "AMS reader",
    "mdl" : "ESP8266",
    "sw" : "2.0.0",
    "mf" : "AmsToMqttBridge"
  }
}`

mqtt topic: "homeassistant/sensor/78ABCD_activeI/config"
`{
  "name" : "AMS active import",
  "stat_t" : "tele/ams_78ABCD/sensor",
  "uniq_id" : "78ABCD_activeI",
  "dev" : {
    "ids" : [ "78ABCD" ]
  },
  "unit_of_meas" : "W",
  "dev_cla" : "power",
  "sta_cla" : "measurement",
  "frc_upd" : true,
  "val_tpl" : "{{value_json['data']['P']}}"
}`

mqtt topic: "homeassistant/sensor/78ABCD_accumI/config"
`{
  "name" : "AMS accumulated active import",
  "stat_t" : "tele/ams_78ABCD/sensor",
  "uniq_id" : "78ABCD_accumI",
  "dev" : {
    "ids" : [ "78ABCD" ]
  },
  "unit_of_meas" : "kWh",
  "dev_cla" : "energy",
  "sta_cla" : "total_increasing",
  "val_tpl" : "{{value_json['data']['tPI'] | default('', true)}}"
}`

There's something missing with the accumulated data settings as I cannot get it to register with the Energy dashboard.

I used this earlier in the sensor.yaml
```
- platform: mqtt
  unique_id: "emeter_active_power"
  state_topic: "tele/ams_78ABCD/sensor"
  name: "Active power"
  unit_of_measurement: W
  device_class: power
  state_class: measurement
  value_template: "{{ value_json.data.P }}"
  json_attributes_topic: "sensor/emeter"
  json_attributes_template: "{{ value_json.data }}"

- platform: mqtt
  unique_id: "emeter_reactive_power"
  state_topic: "tele/ams_78ABCD/sensor"
  name: "Reactive power"
  unit_of_measurement: W
  device_class: power
  state_class: measurement
  value_template: "{{ value_json.data.Q }}"

- platform: mqtt
  unique_id: "emeter_active_energy"
  state_topic: "tele/ams_78ABCD/sensor"
  device_class: energy
  state_class: total_increasing
  name: "Active energy"
  unit_of_measurement: kWh
  value_template: "{{ value_json.data.tPI }}"
```

bardahlm, 2021-12-17:
I have the following in my configuration:

```
   - platform: mqtt
     state_topic: "power"
     name: "power consumtion"
     device_class: "power"
     state_class: "measurement"
     unique_id: "<ID>"
     unit_of_measurement: "W"
     value_template: "{{ value_json['data']['P'] }}"

   - platform: mqtt
     state_topic: "power"
     name: "power meter"
     device_class: "energy"
     state_class: "total_increasing"
     unique_id: "<ID>_tPI_"
     unit_of_measurement: "kWh"
     value_template: "{{ value_json['data']['tPI'] }}"
```
In HA I then get the following:
[image]

bardahlm, 2021-12-17:
I played a bit and for MQTT Discovery I think the following is needed

1. Data should be published to three different topics (one for each list)
2. For every measurement a config must be published following the HA MQTT Discovery standard.

Example for power measurement:

Configuration topic:

`homeassistant/sensor/<mac address>/P/config`

Payload:

`{"device_class": "power", "name": "<meter id> power", "state_topic": "<topic for list 2>", "unit_of_measurement": "W", "value_template": "{{ value_json.data.P}}" }`

Home Assistant will then start putting data from data.P into a sensor called "meter id power" with UOM set to "W".

Voltage U1:

Configuration topic:

`homeassistant/sensor/<mac address>/U1/config`

Payload:

`{"device_class": "voltage", "name": "<meter id> U1", "state_topic": "<topic for list 2>", "unit_of_measurement": "V", "value_template": "{{ value_json.data.U1}}" }`

Accumulated active import:
Configuration topic:

`homeassistant/sensor/<mac address>/tPI/config`

Payload:

`{"device_class": "voltage", "name": "<meter id> tPI", "state_topic": "<topic for list 3>", "unit_of_measurement": "V", "value_template": "{{ value_json.data.tPI}}" }`

To be defined: proper naming of the sensors and possibly giving them an unique ID.

gskjold, 2021-12-18:
Good input guys. It sounds to me like there should be a new choice in the payload type dropdown for this, which also means that the data can be sent as whatever is best suited for HA.

bardahlm, 2021-12-18:
Maybe also add options to select what measurements should be announced. Myself I'd put all the measurements in an Influx database and only some in HA as HA do not really need to know all the measurements.

aadnehovda, 2021-12-20:
I've added some sections for the device information as well, both for the HAN reader
`homeassistant/sensor/hanreader-ed6e_rssi/config`
```json
{
  "name": "WiFi Signal",
  "stat_t": "homeassistant/sensor/hanreader-ed6e/rssi",
  "device_class": "signal_strength",
  "unit_of_meas": "dBm",
  "uniq_id": "hanreader-ed6e_rssi",
  "dev": {
    "identifiers": [
      "hanreader-ed6e", 
      "00000000ED6E"
    ],
    "name": "HAN Reader",
    "model": "PoW-K",
    "sw_version": "2.0.1",
    "manufacturer": "amsleser.no",
    "configuration_url": "http://192.168.1.119"
  }
}
```
and for the AMS meter (notice the `via_device`)
`homeassistant/sensor/ams-6153_importing-active-power/config`
```json
{
  "name": "Importing Active Power",
  "stat_t": "homeassistant/sensor/hanreader-ed6e/meter/import/active",
  "unit_of_meas": "W",
  "device_class": "power",
  "uniq_id": "ams-6153_importing-active-power",
  "dev": {
    "ids": [
      "ams-6153",
      "0000000000006153"
    ],
    "name": "AMS Smart Meter",
    "manufacturer": "Kamstrup",
    "model": "6841121BN245101040",
    "via_device": "hanreader-ed6e"
  }
}
```
As long as there's at least one definition of the device, you can reference it by ID in other sensors:
`homeassistant/sensor/ams-6153_imported-active-energy/config`
```json
{
  "name": "Imported Active Energy",
  "stat_t": "homeassistant/sensor/hanreader-ed6e/meter/import/active/accumulated",
  "unit_of_meas": "kWh",
  "device_class": "energy",
  "state_class": "total_increasing",
  "uniq_id": "ams-6153_imported-active-energy",
  "icon": "mdi:transmission-tower-import",
  "dev": {
    "ids": [
      "0000000000006153"
    ]
  }
}
```
The rest of the sensors for export, reactive, voltage, current, etc would be similar.

aadnehovda, 2021-12-20:
It seems to be a good idea to also add `object_id` to the config blocks to give the sensors well-defined IDs, otherwise Home Assistant will compose a sensor ID based off of the name (and apparently use the `unique_id` under the hood to separate sensors with clashing names). I just used the same value for `object_id` as for `unique_id`, like this
```json
{
  "name": "VCC",
  "stat_t": "homeassistant/sensor/hanreader-ed6e/vcc",
  "device_class": "voltage",
  "unit_of_meas": "V",
  "unique_id": "hanreader-ed6e_vcc",
  "object_id": "hanreader-ed6e_vcc",
  "dev": {
    "ids": [
      "00000000ED6E"
    ]
  }
}
```
[image]

aadnehovda, 2021-12-20:
The structure of MQTT topics may look something like this, where the measure value is posted to the `homeassistant/sensor/<device>/<entity>` topic and the discovery payload to `config`.

Not sure about the names, I see that on the web frontend there's now `instant` and `total`, but `power` and `energy` is perhaps more in line with `current` and `voltage`.

```
homeassistant/sensor/han-ed6e/rssi/config
homeassistant/sensor/han-ed6e/vcc/config
homeassistant/sensor/han-ed6e/free_mem/config
homeassistant/sensor/han-ed6e/uptime/config

homeassistant/sensor/ams-6153/import-power/config
homeassistant/sensor/ams-6153/import-energy/config
homeassistant/sensor/ams-6153/export-power/config
homeassistant/sensor/ams-6153/export-energy/config

homeassistant/sensor/ams-6153/import-reactive-power/config
homeassistant/sensor/ams-6153/import-reactive-energy/config
homeassistant/sensor/ams-6153/export-reactive-power/config
homeassistant/sensor/ams-6153/export-reactive-energy/config

homeassistant/sensor/ams-6153/l1-voltage/config
homeassistant/sensor/ams-6153/l2-voltage/config
homeassistant/sensor/ams-6153/l3-voltage/config

homeassistant/sensor/ams-6153/l1-current/config
homeassistant/sensor/ams-6153/l2-current/config
homeassistant/sensor/ams-6153/l3-current/config

homeassistant/sensor/ams-6153/l1-powerfactor/config
homeassistant/sensor/ams-6153/l2-powerfactor/config
homeassistant/sensor/ams-6153/l3-powerfactor/config
```

gskjold, 2021-12-21:
Very good, this is exactly the kind of information I need. With Christmas and everything I am not sure when I have time to do this, but I'm adding this for 2.1.0

kng, 2021-12-21:
I'm not sure the topic for the actual data should be in homeassistant/#, that's only for the discovery prefix?
sensor/# or amsreader/# would probably be better suited for the data itself.

aadnehovda, 2021-12-21:
Good points. It makes sense to keep the payload as a single block on the same topic, since it originates from the AMS meter as one packet anyway. There won't  be other significant updates between packets. How much of a performance difference it makes to have a single vs many topics I guess may depend on how smart the MQTT library is when flushing the message queue over the wire.

What about just keep using the existing JSON payload (or raw for that matter) and just overlay the extra homeassistant config topics. Whatever publish topic you've chosen would automatically be injected as `state_topic` in the config blocks. A single checkbox on the MQTT config page could then be enough to turn on Home Assistant discovery for either output format, (including the minimal format which is probably enough for most.) If the data payload is JSON then all the config blocks need a `value_template` too of course. 

Another small thing: at least for the full MQTT format the accumulated values are updated only every 60 minutes, while on the web page you can see consumed energy in real time. That would be nice to have for MQTT too. Perhaps the JSON format already does that, I have not tested.

bardahlm, 2021-12-21:
I support having Home Assistant Discovery as an extra option for whatever format being posted to MQTT. I need the data in other applications as well, not only HA.

kng, 2021-12-21:
I'm slowly working on an implementation in my [forked branch](https://github.com/kng/AmsToMqttBridge/tree/homeassistant)
It is very hacky and will need a good cleanup afterwards (I'm no proper c++ coder), but just testing it out and trying to implement the different pieces.

kng, 2021-12-22:
So, after a lot of testing I have a working concept. Still very hacky and hardcoded values. See link in message above.
Separate topics for the different contents, split into /state for rssi, mac etc. /power for the instantaneous values like watts, volts, current. /energy for the accumulated values of kWh. /temperature for temps and /prices for - you guessed it - electrical prices.
All topics have a fixed list of variables to avoid problems.
The /state and /config are run every n times related to /power, so if the meter sends data every 10sec the /state is 30s and /config every 10min. 
It has no retains as for now, nor any other parameters I didn't need to get it up and running. There's probably a lot of room for tweaking.

kng, 2021-12-23:
> Another small thing: at least for the full MQTT format the accumulated values are updated only every 60 minutes, while on the web page you can see consumed energy in real time. That would be nice to have for MQTT too. Perhaps the JSON format already does that, I have not tested.

I have pondered this and it will need to be filtered so it only adds to the reported energy counts, otherwise it will make huge jumps from 0.xkWh to the meter reading, ratcheting up the total_increasing stuff. If this is solved I think it could report the estimated value every 10min or so. HA Energy will probably not use the increased resolution as it's already calculated per hour it seems.

kng, 2021-12-23:
> and for the AMS meter (notice the `via_device`)

Not sure why they should be separated at all. One identifier, like hostname should suffice ? Perhaps also the mac, someone more experienced with keeping an installation working over a long time and swapping out hardware etc, could explain the benefits.

kng, 2021-12-23:
This is what I have built over the last couple of days. I could create a pull, but to a separate branch ? Or how should a big change like this be handled ?
[image]
[image]
[image]

gskjold, 2021-12-24:
Create a pull request to master. As repo owner I have access to change the target to a branch if I want.

gskjold, 2022-01-05:
Sorry I have not yet had much time to look at this. Priorities on current bugs and then release of v2.0.3. Will get back to it at a later point

agekristian, 2022-02-14:
kng

can you preview the code you used?

gskjold, 2022-02-14:
His code have been pulled into branch pr-172. Attached is firmware based on v2.0.10 with the changes he provided in that PR.

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/8060502/esp32.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8060503/esp8266.zip)

agekristian, 2022-02-14:
Hi
I meant the code he used in Home Assistant to get ESP32 up as a device in HA

gskjold, 2022-02-14:
Aha, I guess only he can answer that. I have no experience with HA

kng, 2022-02-14:
> Hi I meant the code he used in Home Assistant to get ESP32 up as a device in HA

The main point of this PR is to require only the discovery option to be enabled in the MQTT integration.
It will then appear as a device with all the sensors configured, just add the sensors to a dashboard.

[image]

agekristian, 2022-02-14:
must I use a special format on "Publish topic" from the AMS reader ?

agekristian, 2022-02-14:
Is it possible that you are another version of the software, because I do not have the option "Home-Assistant" under "Payload"
[image]

kng, 2022-02-14:
That looks like the problem, yes. That feature was one of the first steps I implemented.
I'm not sure you are running the correct version, or if that version has all the changes in it.

gskjold, 2022-02-15:
@agekristian It has not been released yet, hence the zip files in my earlier post containing a version with this feature.

libo, 2022-02-15:
I am also on HA. I am giving it a QA spin on a ESP32.
@gskjold anything specific you want us to to test beside the integration works as expected?

libo, 2022-02-15:
Looks like I am getting the config! Awesome

Getting all the config topic (but still no reading, example rssi)
[image]

[image]

My problem right now is that I still have to figure out if PUSH is active in my Danish HAN so I still don't get any reading (working with Radius)

That said, does the Bridge publish rssi independently of reading coming in from the HAN interface?

libo, 2022-02-15:
Got my PUSH activated and... tada!

this is a neat feature for Home Assistant people!
[image]

It also integrates seamlessly with the Energy pane in HA
[image]

Mind blowing! 
Thanks a lot @kng !

gskjold, 2022-02-15:
Good stuff! I guess there is a general issue that system information like rssi is not published if there is no data from the meter, I will look into that.

libo, 2022-02-15:
I also managed to push the MQTT messages that HomeAssistant receives into a [InfluxDB Cloud ](https://www.influxdata.com/products/influxdb-cloud/) (free plan)

This is a quick dashboard I put together in few minutes

[image]

Here is what to add to HomeAssistant `configuration.yml`:
```

influxdb:
  api_version: 2
  ssl: true
  host: eu-central-1-1.aws.cloud2.influxdata.com # Or whatever you end up using in term of computing and region
  token: "<your token>" # once in influxdata.com, left menu 'Load data' -> API Tokens
  organization: '<email you used to register on influxdata.com>'
  bucket: <name you gave to the bucket on influxdata.com>
  tags:
    source: HA
  tags_attributes:
    - friendly_name
  default_measurement: units
  include:
    entity_globs:
      - sensor.<The Client ID in the MQTT setup in AMS reader>* # in my case Client ID is 'elcounter' so I used 'sensor.elcounter*'

```

Shall we add this stuff to the Wiki?

🪄

ArnieO, 2022-02-15:
> Shall we add this stuff to the Wiki?

Yes, please! 👍

bardahlm, 2022-02-15:
Can this feature co-exist with json payload?

kng, 2022-02-17:
> Can this feature co-exist with json payload?

Posting both at the same time ? No. 
Although both json/HA use json payload, it would be quite easy to use the json fom this HA type as it is a bit stricter and still contain all the same fields. they're split into several different topics for this reason.

enordli, 2022-03-07:
Daniel, i deployd your branche pr-172 and it worked stright forward except that values for volt, current, reactive export and reactive import are cleared out until next 10 sec payload.
What did you do with MQTT otions in HA to avoid that ?

[image]

svhaaland, 2022-03-08:
Had same issue with my Aidon meter.
Changed following in the branche pr-172 to get this to work with HA.
https://github.com/svhaaland/AmsToMqttBridge/commit/21a7adec49b0089fa3ef2fdce2a1ec1160bab019
[image]
[image]

enordli, 2022-03-08:
Thank You,
Now it's working as expected.

bjornsivertsen, 2022-03-16:
Hi, I am a bit confused how to get the auto discovery working.

I am running the latest beta, that has the home-assistant payload possibility.

[image]

I am also getting the messages, which I can read if I listen to the topic through the MQTT integration:

[image]

But no device is created, even though auto discovery is enabled.

[image]

bjornsivertsen, 2022-03-16:
Hi, I am struggling a bit with getting the auto-discovery to work.

I am running the latest beta, and have the MQTT messages sent to my HA Mosquito Broker Supervisor addon, which works fine:

[image]
[image]

I am also able to listen in on a topic from HA, for instance: homeassistant/sensor/kamstrup-meter/config/energy

[image]

But no devices are created, through auto-discovery, even though it is enabled:

[image]

Anyone got any pointers?

kng, 2022-03-16:
I don't think it will work when publishing the sensor data in to the discovery topic.
Try setting the topic to: kamstrup-meter or perhapskamstrup-meter/energy ?

enordli, 2022-03-16:
> I did as what you said, and some stuff did start appearing in the the logs, though no device. :( And it appears that the logs are related to some AMS stuff published to another topic that I didn't specify, as it can be seen in the explorer.
> 
> [image] [image]
> 
> [image]
> 
> Logger: homeassistant.components.mqtt Source: components/mqtt/**init**.py:1115 Integration: MQTT (documentation, issues) First occurred: 19:59:11 (34 occurrences) Last logged: 20:02:04
> 
> ```
> Can't decode payload b'{"name":"L3 voltage","stat_t":"\x04\x8d\xff?\x0f/power","uniq_id":"ams-c23f_U3","obj_id":"ams-c23f_U3","unit_of_meas":"V","val_tpl":"{% if value_json.U3 is defined %} {{ value_json.U3 }} {% else %} {{(states.sensor.ams-c23f_U3|float)}} {% endif %}","dev_cla":"voltage","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"53296cc","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":null}' on homeassistant/sensor/ams-c23f_U3/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f753a1ee0>>)
> Can't decode payload b'{"name":"Accumulated active import","stat_t":"\x04\x8d\xff?\x0f/energy","uniq_id":"ams-c23f_tPI","obj_id":"ams-c23f_tPI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tPI is defined %} {{ value_json.tPI }} {% else %} {{(states.sensor.ams-c23f_tPI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"53296cc","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tPI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f753a1ee0>>)
> Can't decode payload b'{"name":"Accumulated active export","stat_t":"\x04\x8d\xff?\x0f/energy","uniq_id":"ams-c23f_tPO","obj_id":"ams-c23f_tPO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tPO is defined %} {{ value_json.tPO }} {% else %} {{(states.sensor.ams-c23f_tPO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"53296cc","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tPO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f753a1ee0>>)
> Can't decode payload b'{"name":"Accumulated reactive import","stat_t":"\x04\x8d\xff?\x0f/energy","uniq_id":"ams-c23f_tQI","obj_id":"ams-c23f_tQI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQI is defined %} {{ value_json.tQI }} {% else %} {{(states.sensor.ams-c23f_tQI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"53296cc","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f753a1ee0>>)
> Can't decode payload b'{"name":"Accumulated reactive export","stat_t":"d\x83\xff?\x0f/energy","uniq_id":"ams-c23f_tQO","obj_id":"ams-c23f_tQO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQO is defined %} {{ value_json.tQO }} {% else %} {{(states.sensor.ams-c23f_tQO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"53296cc","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7f753a1ee0>>)
> ```

To listen on HA for sensor configuration you have to use this string: homeassistant/sensor/#
and it will look like this if is ok.
[image]

if got nothing try to restart AMS reader when you are listening

If it does not work, then you should try to delete the Mosquitto broker, do a restart and add it again.
The only thing you have to add is the ip and port  of MQTT server,
to force AMS reader to send the configuration MQTT, do a restart of the AMS reader

kng, 2022-03-16:
That looks quite odd, but perhaps it points to some special character got in to the topic string.
The old discovery stuff should be removed from mqtt retain as well.
Also, HA only accepts one discovery per startup, so a restart under Configurtion -> Server Controls is needed.

enordli, 2022-03-16:
What i did was that i tested on a HA i use for testing, and after i deleted the MQTT broker, restarted and added it again with the ip and port, the 1 device and 17 entities did not show up before i restarted the AMS reader. i dont know why, but that desent matter.
(I did not delete the entities before i deleted MQTT broker) 
if it will for some reason be conflict whith the devices then rename the hostname on the AMS reader, and you will get new sensor names.

bjornsivertsen, 2022-03-17:
Still struggling sadly.

What I have done:
1. Kept kamstrup-meter as the publish topic and not homeassistant/sensor/kamstrup-meter
2. Restarted my HA instance
3. Started listening on homeassistant/sensor/#
4. Restarted AMS reader

No messages gets picked up by HA, but if I browse the broker through MQTT Explorer, I still get the readings, as well as the odd messages passed to homeassistant/sensor that I did not define anywhere.

[image]

They get sent on AMS startup, so are they the discovery messages?

If I listen in on another topic, for instance kamstrup-meter/energy, I can see the messages through HA, so I doubt it's an access issue?

kng, 2022-03-17:
This is becoming a educational topic on how HA works, I'm not sure that is the intended purpose. Read the [docs.](https://www.home-assistant.io/docs/mqtt/discovery/)
The homeassistant/* is the discovery part, it should only get the description of what messages to subscribe to and how to parse them. in this case they're sent once and retained, which mean each time HA is restarted it gets the info.
As long as you don't change the hostname (ams-c23f here) the changes in the discovery will be overwritten, and only need a HA restart.
In the topic you specified there will be three sub topics published with the actual data(sensors), energy/power/state.
In this case it is visible you made the change to reduce the topic lenght so it shows up in two places.
You should look in the Configuration -> Devices where the reader should appear if the discovery went well, then check out the MQTT INFO button to see what it has gotten. 
[image]

bjornsivertsen, 2022-03-17:
I am simply guessing here, but it could be related to the logs I posted before. Where MQTT can't decode the discovery message, and simply doesn't show them in the listening page. And thus no device is created through auto-discovery.

The issue might be related to the errors described around here, related to 8266 boards and Kamstrup setups.

https://github.com/gskjold/AmsToMqttBridge/issues/247#issuecomment-1069148359

A fix has already been found, and I will test again when it has been implemented.

bjornsivertsen, 2022-03-17:
Unfortunately, after cleaning everything up and applying the fix that impacted my specific setup, I still get errors related to the discovery message.

From HA logs:
`Logger: homeassistant.components.mqtt
Source: components/mqtt/__init__.py:1115
Integration: MQTT (documentation, issues)
First occurred: 19:13:04 (34 occurrences)
Last logged: 19:13:04

Can't decode payload b'{"name":"Accumulated active export","stat_t":"\xc4r\xff?\x0f/energy","uniq_id":"ams-c23f_tPO","obj_id":"ams-c23f_tPO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tPO is defined %} {{ value_json.tPO }} {% else %} {{(states.sensor.ams-c23f_tPO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tPO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function websocket_subscribe.<locals>.forward_messages at 0x7f9e5ec550>>)
Can't decode payload b'{"name":"Accumulated reactive import","stat_t":"\xc4r\xff?\x0f/energy","uniq_id":"ams-c23f_tQI","obj_id":"ams-c23f_tQI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQI is defined %} {{ value_json.tQI }} {% else %} {{(states.sensor.ams-c23f_tQI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated reactive import","stat_t":"\xc4r\xff?\x0f/energy","uniq_id":"ams-c23f_tQI","obj_id":"ams-c23f_tQI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQI is defined %} {{ value_json.tQI }} {% else %} {{(states.sensor.ams-c23f_tQI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function websocket_subscribe.<locals>.forward_messages at 0x7f9e5ec550>>)
Can't decode payload b'{"name":"Accumulated reactive export","stat_t":"\xc4r\xff?\x0f/energy","uniq_id":"ams-c23f_tQO","obj_id":"ams-c23f_tQO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQO is defined %} {{ value_json.tQO }} {% else %} {{(states.sensor.ams-c23f_tQO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated reactive export","stat_t":"\xc4r\xff?\x0f/energy","uniq_id":"ams-c23f_tQO","obj_id":"ams-c23f_tQO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQO is defined %} {{ value_json.tQO }} {% else %} {{(states.sensor.ams-c23f_tQO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function websocket_subscribe.<locals>.forward_messages at 0x7f9e5ec550>>)`

bjornsivertsen, 2022-03-17:
Ok, I have cleaned up messages, reinstalled the MQTT broker, and applied the fix and sadly I am still getting the errors in the HA log related to the discovery message when I restart the AMS reader.

[image]
[image]
[image]

```
Logger: homeassistant.components.mqtt
Source: components/mqtt/__init__.py:1115
Integration: MQTT (documentation, issues)
First occurred: 19:13:04 (119 occurrences)
Last logged: 19:51:44

Can't decode payload b'{"name":"L3 voltage","stat_t":"\x8c\x85\xff?\x0f/power","uniq_id":"ams-c23f_U3","obj_id":"ams-c23f_U3","unit_of_meas":"V","val_tpl":"{% if value_json.U3 is defined %} {{ value_json.U3 }} {% else %} {{(states.sensor.ams-c23f_U3|float)}} {% endif %}","dev_cla":"voltage","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":null}' on homeassistant/sensor/ams-c23f_U3/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated active import","stat_t":"\x8c\x85\xff?\x0f/energy","uniq_id":"ams-c23f_tPI","obj_id":"ams-c23f_tPI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tPI is defined %} {{ value_json.tPI }} {% else %} {{(states.sensor.ams-c23f_tPI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tPI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated active export","stat_t":"t\x85\xff?\x0f/energy","uniq_id":"ams-c23f_tPO","obj_id":"ams-c23f_tPO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tPO is defined %} {{ value_json.tPO }} {% else %} {{(states.sensor.ams-c23f_tPO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tPO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated reactive import","stat_t":"t\x85\xff?\x0f/energy","uniq_id":"ams-c23f_tQI","obj_id":"ams-c23f_tQI","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQI is defined %} {{ value_json.tQI }} {% else %} {{(states.sensor.ams-c23f_tQI|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQI/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
Can't decode payload b'{"name":"Accumulated reactive export","stat_t":"t\x85\xff?\x0f/energy","uniq_id":"ams-c23f_tQO","obj_id":"ams-c23f_tQO","unit_of_meas":"kWh","val_tpl":"{% if value_json.tQO is defined %} {{ value_json.tQO }} {% else %} {{(states.sensor.ams-c23f_tQO|float)}} {% endif %}","dev_cla":"energy","dev":{"ids":["ams-c23f"],"name":"AMS reader","mdl":"ESP8266","sw":"beafcd3","mf":"AmsToMqttBridge","cu":"http://ams-c23f.local/"},"stat_cla":"total_increasing"}' on homeassistant/sensor/ams-c23f_tQO/config with encoding utf-8 (for <Job HassJobType.Coroutinefunction <function async_start.<locals>.async_discovery_message_received at 0x7faa7a5550>>)
```

gskjold, 2022-03-17:
Interesting, it seems that the topic part in "stat_t" gets corrupted when it is over a certain length. Try to have only "kamstrup" as publish topic in MQTT config

gskjold, 2022-03-17:
Found a fix, will publish new version in #247  soon

bjornsivertsen, 2022-03-17:
Success!

[image]

kng, 2022-03-18:
As I've stated earlier, using the discovery topic for data is risky, it could be the case that one of them is consumed by the discovery process and therefore isn't available for the data.
Please try using a topic outside homeassistant/#
There should be useful info in the logs as well.
State class does not need to be set, only if you need certain functionality to be added.
https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes

erlandp, 2022-03-18:
Thank you for clarifying regarding stste classes. To be honest I had not read up on that.
I'm not sure what you mean by the "using the discovery topic for data" though?

I had another try, my HA does not like the 'null' in 'stat_cla: null'. If I resend the message with the line 'stat_cla: null' omitted  or edited to 'stat_cla: measurement' everything works as expected.

gskjold, 2022-03-18:
@kng have you tested the latest in #247 ? I did make some adjustments to your original implementation. Maybe I broke something :) If omitting stat_cla instead of setting it as null (which I think was my adjustment), I will fix that.

[7 more replies]

closed 2022-03-30 by gskjold