UtilitechAS/amsreader-firmware #162: Power factor not included in MQTT

henriklund, 2021-12-12:
Super stuff done on the new v2.0.0!!!

I can see that PowerFactor is now included in the JSON. However, when querying MQTT it does not appear to be included.

Deployed version is ams2mqtt-esp8266-2.0.0.bin on a Pow-K module.

henriklund, 2021-12-13:
Duly noted. I will look forward to 2.0.2 then :)

madsbg, 2021-12-13:
What is power factor and how do I know if my meter supports it?

I use MQTT Raw (full) format, and I receive {root}/meter/powerfactor and {root}/meter/l1/powerfactor, which both have a value of zero, but neither {root}/meter/l2/powerfactor nor {root}/meter/l3/powerfactor are present.

henriklund, 2021-12-14:
You can read about power factor [here](https://www.fluke.com/en/learn/blog/power-quality/power-factor-formula). As for whether your meter is supported, I guess it might depend on model. Mine is the Kamstrup Omnipower [DIN rail](https://www.kamstrup.com/en-en/electricity-solutions/meters-devices) version.

Do you get the same results if you query http://meter-ip/data.json? (look for the f + f1-f3 values).

madsbg, 2021-12-14:
I have the Kamstrup Omnipower three-phase meter (Denmark/SEAS-NVE).

data.json:
[image]

MQTT:
[image]

henriklund, 2021-12-14:
My guess is that it is either a bug (@gskjold could elaborate here) that causes data not to get loaded from meter or that your meter might not have the firmware necessary for this (see the link in [this post](https://github.com/gskjold/AmsToMqttBridge/issues/136)).

madsbg, 2021-12-14:
Thanks. What does your data.json look like?

henriklund, 2021-12-14:
Redacted some irrelevant info:
[image]

gskjold, 2021-12-16:
#163 will at least fix what is missing in MQTT raw

gskjold, 2021-12-16:
Commited fix

closed 2021-12-16 by gskjold