UtilitechAS/amsreader-firmware #165: Missing tPI value in MQTT-message

thrawnarn, 2021-12-17:
I'm missing the tPI value in my json mqtt-message. And I can see in the wiki that there is 3 kinds of lists in the mqtt message. How do I choose which to pubish?

https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats

gskjold, 2021-12-17:
The different lists are based on what the meter provides. Please provide country and meter model.

thrawnarn, 2021-12-17:
I live in Sweden and has a Kamstrup meter. Do you need to know the exakt model?

thrawnarn, 2021-12-17:
I can see the values in the web-interface

[image]

gskjold, 2021-12-17:
Not sure how it is with swedish kamstrup. If it uses the norwegiam standard (as some swedish meters do), you will only receive list3 once every hour.

thrawnarn, 2021-12-17:
hmm, ok. I think mine is using the norwegian standard. Then I will wait an hour to see if it updates :)
Thanks

closed 2021-12-17 by thrawnarn

gskjold, 2021-12-17:
No problem. You should get it approx 23.00.10

thrawnarn, 2021-12-17:
FYI: I've confirmed it in Home Assistant now, I could see it in the history. Now I will have to find a way around this... But thats tomorrows problem.

gskjold, 2021-12-17:
Yeah, pretty inconvenient... I calculate an real time estimate in openhab based on current active import and time since last message. It is not exact, but good enough for me. Same estimate calculation is used on accumulated values in gui. You will see they get corrected (forward or backward) when you receive list 3