UtilitechAS/amsreader-firmware #171: v2.0.2 minor issues

ArnieO, 2021-12-20:
**Describe the bug**

1. When upgrading from 2.0.1,  daily consumption historic was lost (hourly not lost)
2. The clock is not correct:
[image]
Current date/time: 20th December 2021, 11:18

**Hardware information:**
 - Meter: Kamstrup
 - Pow-K

**Relevant firmware information:**
 - Version: 2.0.2
 - MQTT: No
 - HAN GPIO: UART0
 - Temperature sensors: No

gskjold, 2021-12-20:
Thanks, that's interesting. I will look in to what may cause month graph to be cleared. I'm guessing it is related to the clock,

Judging by the timestamp your device have been up for 12 minutes at that point and should have been updated. It suggests that NTP does not work, either because of configuration (NTP config or missing DNS in IP config) or firewall.

If NTP is not possible in your setup, the clock will synchronise with meter timestamp at next hourly update.

ArnieO, 2021-12-20:
I'm in a new home and wifi conditions are not good (many other things to take fix...). Maybe that could cause the NTP issue?

But after two hours the clock is still not synchronized:
[image]
[image]

ArnieO, 2021-12-20:
Correction: 
When I lost daily values I did not upgrade from 2.0.1 but from 2.0-rc4.
That may explain it.

gskjold, 2021-12-21:
Ah yes, there was a problem with the earlier rc of 2.0

ArnieO, 2021-12-21:
The clock seems to run, but the date is stuck. 
[image]

bardahlm, 2021-12-21:
Can the timestamps from the meter be used if NTP is not working?

gskjold, 2021-12-21:
It should use meter timestamp if NTP is not available. There must be something wrong since that doesn't happen

gskjold, 2021-12-21:
@ArnieO could you screenshot your ntp config page and ip config page? Also, are you using static IP? If so, what DNS have you configured?

ArnieO, 2021-12-21:
I have not specified DNS, as my understanding was that it is only needed for SSL. Wrong?

[image]
[image]

gskjold, 2021-12-21:
Needed for NTP and ENTSO-E to work. It is unable to look up pool.ntp.org

I have also found out why it doesn't use meter time

gskjold, 2021-12-22:
> my understanding was that it is only needed for SSL. Wrong?

Not entirely untrue. DNS is needed for SSL because SSL requires a synchronized clock which we would need to get from NTP. Meter timestamp would of course also work, but then you have to wait for the next hourly update before it works.

closed 2021-12-27 by gskjold