UtilitechAS/amsreader-firmware #173: Kamstrup, Denmark, Invalid HDLC retured with -9 error

GGBMLC, 2021-12-25:
Hi
My D1 mini board seems to work fine with the setup. 
I have add the encryptions.
When I connect the setup to my Meter then first "HAN" is green, then change to Orange and then change to RED. 

Telenet log shows:
(readHanPort) Frame dump (10b):
85 50 DE 07 0E 15 BB 9B 5D D0 (this change for ever new reading)
(readHANPort) Invalid HDLC, returned with -9

I don't know what the returned with -9 error is indicate?

It could be a suggestion to write the error information on the webinterface.

**Hardware information:**
 - Meter: Kamstrup, 3F, Denmark
 - AMS reader: Wemos ESP8266 D1 mini 
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.02
 - MQTT: [yes/no] , not tested as I can not get dato from the meter
 - HAN GPIO: GPIO5. Data_out pin 5 is connected to pin "D1" on the D1 mini board
 - Temperature sensors Not used/tested
 -
[image]

gskjold, 2021-12-25:
There have been a couple of similar cases lately which have been solved by selecting UART2 in GPIO setting and connecting HAN to pin 13. I am not sure what caused this to be a problem in the later versions, but I should obviously have a look at it.

GGBMLC, 2021-12-25:
Hi 
Thank you for feedback. 
I try with the HAN set to UART2 (GPIO 13)  , Data out from the meter to D2, D1 mini board. 
Still same error. 
[image]

gskjold, 2021-12-25:
Im assuming you hooked it up to D7 (gpio13)

I cant really see why it should receive 10b chunks, I'll have to think about how that could happen.

gskjold, 2021-12-27:
Please try the attached firmware
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/7779846/esp8266.zip)

GGBMLC, 2021-12-27:
Hi
I tried the new version. 
Still same error. 
(readHANPort) Invalid HDLC, returned with -9

Note: GPIO set to UART2 (GPIO13)
[image]

GGBMLC, 2021-12-27:
Hi 
I tried with different meter setting. IF I change to "Parity 8N1" I was able to read some of the data from the Meter.
Then with a returned error "-1"

[image]
[image]

gskjold, 2021-12-27:
8N1 is correct for Kamstrup. 491 bytes is correct length for what is specified by header (0x1E9 = 489, +2 frame boundary tags), but the last byte is not 0x7E as expected. Does this happen for all frames?

GGBMLC, 2021-12-27:
Hi
I look at the log one more  time. 
Each frame from the Kamstrup meter is 491.  But the data is not end with "0X7E.

GGBMLC, 2021-12-27:
[image]

ArnieO, 2021-12-27:
I consistently receive 228b from my Kamstrup:
[image]

gskjold, 2021-12-27:
Danish Kamstrup with encryption is always 491 as far as I know. Matches other Danish frames I have kept for reference earlier.

gskjold, 2021-12-27:
> [image]

Third row from the bottom you will find 7E 7E which is end of frame and then start of the next frame. The payload still suggests 491 bytes, so it looks like you have packet loss.

@ArnieO is it valid to hook GPIO directly to DATA_OUT on Kamstrup? Is it correct voltage?

GGBMLC, 2021-12-27:
[5512-2584_GB_B1_OMNIPOWER HAN interface specification.pdf](https://github.com/gskjold/AmsToMqttBridge/files/7780516/5512-2584_GB_B1_OMNIPOWER.HAN.interface.specification.pdf)
[DLMS-COSEM.pdf](https://github.com/gskjold/AmsToMqttBridge/files/7780518/DLMS-COSEM.pdf)

ArnieO, 2021-12-27:
> Third row from the bottom you will find 7E 7E which is end of frame and then start of the next frame. The payload still suggests 491 bytes, so it looks like you have packet loss. 
> @ArnieO is it valid to hook GPIO directly to DATA_OUT on Kamstrup? Is it correct voltage?

Yes, DATA_OUT is 3,3V data signal that can be connected directly to ESP GPIO. This is what I do on Pow-K.

Could it be a brownout issue? 
I always suspect insufficient decoupling capacitance on the D1 mini boards. To be reliable the ESP8266 need a good current buffer (capacitance with low equivalent serial resistance) close to its power pins. On Pow-U I have taken it to the extreme by using both a ceramic capacitor 10uF close to the ESP8266 and a low-ESR 1F supercapacitor for the board. The supercap value is probably overkill but is the same component I use on Pow-U (where it is needed) - and this has proven to give a very stable solution. But I would at least try to add a good size ceramic capacitor with short leads close to the ESP power pins.

GGBMLC, 2021-12-27:
I power the D1 mini with a USB mobile charger. I can try to measure the voltage on the board. Maybe check wiht my oscilloscope for voltage drop.

ArnieO, 2021-12-27:
> I power the D1 mini with a USB mobile charger. I can try to measure the voltage on the board. Maybe check wiht my oscilloscope for voltage drop.

This can be hard to detect on an oscilloscope, as the reason is that the ESP8266 draws some short and heavy current spikes during some parts of its wifi transmission cycle. So the voltage drop could be very short - but please try! 
Be careful with the grounding, I would recommend using a battery or power bank, to avoid ground loop (depends on how your USB charger is constructed). Remember, the GND side of the oscilloscope probe is connected to ground in your power socket.
And to get the reading you want, both sides of the oscilloscope probe should be as close as possible to Vcc and GND pins on the ESP8266 module on your D1.

GGBMLC, 2022-01-04:
Hi 
Thank you for support. I have now tried with a new board. Wireless module ESP8266-NodeMcu v3 CH340. 
After some minor settings issue I am now able to get data  from my Kamstrup meter. 
I use HAN, GPIO13.  "D7" on the board. 
Suggest for improvement. Add IP info address in the GUI. (then it is more easy to start the Telnet debug).
 
SW version 2.0.2

ArnieO, 2022-01-04:
> Suggest for improvement. Add IP info address in the GUI. (then it is more easy to start the Telnet debug).

My general recommendation is to use fixed IP, but for those not doing that, this is a great idea! 👍
It is in fact strange that we haven't thought about it earlier...

@gskjold : If you agree to do this, I suggest you place it as read-only information on the Configuration/Wifi page. Maybe also on the System/Debug page, for convenience.

gskjold, 2022-01-05:
Thanks for the input, I agree, will fix, adding #188

closed 2022-01-05 by gskjold